espressif/esp-homekit-sdk #1: lightbulb example not working

holgi, 2020-10-12:
The lightbulb example does not work out of the box.

When defining the `ledc_timer` in `lightbulb.c` (line 169 cont.) I had to add the `duty_resolution` property to get it to work. 

Complete code:
```
    // config the timer
    ledc_timer_config_t ledc_timer = {
        //set frequency of pwm
        .freq_hz = 5000,
        //timer mode,
        .speed_mode = LEDC_HIGH_SPEED_MODE,
        //timer index
        .timer_num = LEDC_TIMER_0,
        // resolution of PWM duty
        .duty_resolution = LEDC_TIMER_13_BIT
    };
```

OS: macOS Catalina 10.15.7
ESP-IDF Version: ESP-IDF v4.3-dev-1197-g8bc19ba89-dirty
esp-homekit-sdk: git commit f2a010a

shahpiyushv, 2020-10-12:
@holgi , thanks for reporting this. I also realised that the reset gpio logic is also interfering with the LED logic because they share the same gpio (no. 0). We will fix this.

oksimple, 2021-06-03:
Lamp example missing definition of LEDC_HIGH_SPEED_MODE