espressif/esp-homekit-sdk #2: garage door

coffice12, 2020-10-15:
绑定提示您可以在生产企业app中允许此配件访问家庭

这个app在哪里呢

Eavo, 2020-10-16:
生产企业 App ，我的理解是产品厂商开发的 App，并启用该 App 访问苹果 Home 的权限。如果你们没有自己的 App 可以忽略并使用苹果原生的 Home App  即可。

coffice12, 2020-10-16:
> 生产企业 App ，我的理解是产品厂商开发的 App，并启用该 App 访问苹果 Home 的权限。如果你们没有自己的 App 可以忽略并使用苹果原生的 Home App 即可。

我就是在home app里面使用并且收到该提示

shahpiyushv, 2020-10-20:
@coffice12 , your problem isn't clear to me. While adding the accessory using home app, you will see a warning as shown here in the screenshot. If you are seeing something different, can you share a screenshot?
[image]

shahpiyushv, 2020-12-23:
Closing this issue as there was no response. Please feel free to reopen if required.

closed 2020-12-23 by shahpiyushv