espressif/esp-homekit-sdk #4: Decryption error/Connection lost on ESP8266

GoRo33, 2020-10-18:
Hi!

I have reported a problem with "data_tlv8" example. When I try to connect to device i getting connection problem at end of paring process: 

`######## Starting Pair Setup ########
Pair Setup M1 Received
Pair Setup M2 Successful
Cleaning Pair Setup Context
######## Starting Pair Setup ########
Pair Setup M1 Received
Pair Setup M2 Successful
Cleaning Pair Setup Context
######## Starting Pair Setup ########
Pair Setup M1 Received
Pair Setup M2 Successful
Pair Setup M3 Received
Using pair-setup without MFi.
Pair Setup M4 Successful
Pair Setup M5 Received
Pair Setup Successful for B5D20CBA-70F8-45C9-BF33-F978E31F656C
Updated state number to 2
Re-announcing _hap._tcp mDNS service
Cleaning Pair Setup Context
Cleaning Pair Setup Context
######## Starting Pair Verify ########
Pair Verify M1 Received
Pair Verify M2 Successful
Pair Verify M3 Received
HomeKit Session active
Pair Verify Successful for B5D20CBA-70F8-45C9-BF33-F978E31F656C
I (45327) HAP Sensor: Received Read for Current Temperature
I (45329) HAP Sensor: Current Value: 
I (45330) HAP Sensor: Updated Value: 
Value Changed
Read data: 
aa 0b 0c 0d 
Value Changed
Read tlv8: 
Value Changed
Decryption error/Connection lost. Marking session as invalid
HomeKit Session terminated
######## Starting Pair Verify ########
Pair Verify M1 Received
Pair Verify M2 Successful
`

Lightbulb example work fine with connection but when i change `cfg` structure to `.cid = HAP_CID_SENSOR,` i getting this same error at connection. 

I have encounter tis error because i try to add humidity and temperature characteristics. 
My os is Ubuntu 18.04 
and sdk version is: commit f2a010a53f937305266e01664b0dece6ab1fa979 (HEAD -> master, origin/master, origin/HEAD)
esp8266_rtos_sdk version is: commit 66dd5798803a96a9f2a7d77099bc36d7bef918dc (HEAD -> master, origin/master, origin/HEAD)

shahpiyushv, 2020-10-20:
@GoRo33 seems some issue with printing float numbers. In order to confirm this, please add a call `hap_http_debug_enable()` before `hap_start()` in your `app_main.c` and check if you see this line in the GET /accessories response

```
"characteristics":[{"iid":12,"value":,"type":"11","perms":["pr","ev"],"ev":false,"format":"float","minValue":,"maxValue":,"minStep":,"unit":"celsius"}
```

As you can see, the values are missing for multiple keys. Since this is not a valid JSON, iOS rejects this.

Please search for the below config option in your sdkconfig file and change it from `y` to `n`.

```
CONFIG_NEWLIB_NANO_FORMAT=y
```

Let me know if it fixes the issue for you.

GoRo33, 2020-10-20:
@shahpiyushv Hi!
Thanks for the help. This solves my problem. As far I know there are some error with printing float numbers on ESP8266 RTOS SDK. When using `printf` or `ESP_LOG`, float format don't work.

Should I this solution be in default config for ESP8266?

shahpiyushv, 2020-10-20:
@GoRo33 , If you mean the sdkconfig.defaults file (and not defaults within ESP8266_RTOS_SDK itself), then yes, atleast for the examples that have floats, this needs to be set.

Mike09, 2021-05-27:
Hi there,

I'm having the same issue while using the lightbulb example.
For my use case I wanted to add some DHT22 sensor to read temperature and humidity from.
In my test, I only used the temperature and passed it to the hap characteristics without updating the temperature later on.
When reading and printing the information from the sensor as int everything works fine.
Additionally, the `CONFIG_NEWLIB_NANO_FORMAT=y` is set by default to yes but unfortunately I cannot print any float values.
I removed line `hap_enable_mfi_auth(HAP_MFI_AUTH_HW);` as I'm not using the MFI variant of sdk.

Any suggestion what I might be missing?
@GoRo33 do you have any example code for your solution?

Thanks in advance!

shahpiyushv, 2021-05-27:
@Mike09 , as you can see in my [comment above](https://github.com/espressif/esp-homekit-sdk/issues/4#issuecomment-712879446), search for `CONFIG_NEWLIB_NANO_FORMAT=y` and change it from y to n. Probably you misread the comment and retained `CONFIG_NEWLIB_NANO_FORMAT=y`?

danieldegrasse, 2021-11-07:
In my experience, disabling `CONFIG_NEWLIB_NANO_FORMAT` resulted in code size being too large for the ESP8266, which produces the `Fatal exception (28)` error. Enabling release optimizations with `CONFIG_OPTIMIZATION_LEVEL_RELEASE=y` will reduce code size sufficiently to fix the problem.

andrewteta, 2022-04-21:
I'm having the same issue on ESP32:
> ######## Starting Pair Verify ########
Pair Verify M1 Received
Pair Verify M2 Successful
Pair Verify M3 Received
I (6198) HAP Lightstrip: Controller 73611AA3-78CF-4BFC-9316-98E46855FED2 Connected
HomeKit Session active
Pair Verify Successful for 73611AA3-78CF-4BFC-9316-98E46855FED2
Decryption error/Connection lost. Marking session as invalid
I (11588) HAP Lightstrip: Controller 73611AA3-78CF-4BFC-9316-98E46855FED2 Disconnected
HomeKit Session terminated

HomeKit SDK commit (master): [eade505a](https://github.com/espressif/esp-homekit-sdk/commit/eade505ade6c148056044307ce7b779bbc8e218b)

ESP-IDF commit (v4.4): [8153bfe412](https://github.com/espressif/esp-idf/tree/8153bfe4125e6a608abccf1561fd10285016c90a)

I've tried the troubleshooting steps that @shahpiyushv suggested above, but this did not work. I also tried commenting out all `ESP_LOGI()` calls with float values - again, same problem. Restarting my modem, router, DHCP server, etc. has not made any difference.

At this point I'm at a loss for what could be going on or how to properly debug it. 

[Here](https://github.com/espressif/esp-homekit-sdk/files/8527071/hap_http_debug.txt) is the output from `hap_http_debug_enable()`

Any help would be much appreciated!

andrewteta, 2022-04-24:
UPDATE:
I was finally able to resolve this issue. Going into the Home app on my iPhone to "**Delete my home**" fixed everything. Once I added my HomeKit device again, it worked on the first try.

Step-by-step instructions for this can be found [here](https://www.macworld.co.uk/how-to/how-fix-homekit-devices-wont-connect-problems-3777667/)