espressif/esp-homekit-sdk #8: Blinds HOMEKIT (servo or stepper motors)

exrector, 2020-11-08:
Hello. I will repeat my question here, because this topic is closer.

> Hello dear developers! Can I ask you to make examples on servo or stepper motors control? Ideally, these are blinds. PLEASE! There is no information on the internet. It is very difficult for a beginner to understand the examples of other platforms (arduino, esp8266 etc)

I used several resources and instructions:
https://github.com/espressif/esp-iot-solution/tree/master/components/motor/servo
https://github.com/maximkulkin/esp-homekit-demo/wiki/Build-instructions-ESP32
https://github.com/espressif/esp-rainmaker/tree/master/examples/homekit_switch

But I cannot merge the files. I can not. I do not have sufficient knowledge. Could you make an example for motors? This project is my dream! My dream is to make smart blinds. For this reason, I started studying your techniques a few months ago.

I have several esp32 boards. 
Stepper motor 28BYJ-48 5V + driver ULN2003. 
Also I have servo motor MG996R 5V + driver L298N

Please make an example HOMEKIT MOTOR for which I might understand this.

shahpiyushv, 2020-11-09:
@exrector , I am not sure if I understand the exact issue here.

- Is this just a problem of merging https://github.com/espressif/esp-iot-solution/tree/master/components/motor/servo into https://github.com/espressif/esp-rainmaker/tree/master/examples/homekit_switch?
- Do you have both of these working fine independently?
- Have you been able to create atleast a dummy HomeKit service for your requirement without the hardware integration? The "Window Covering" is the service you might be looking for. Have you checked that in the HomeKit specifications? It can also be emulated using [this HomeKit Emulator example](https://github.com/espressif/esp-homekit-sdk/tree/master/examples/emulator).

exrector, 2020-11-14:
Hello! @shahpiyushv Thank you for responding
1. Project https://github.com/espressif/esp-rainmaker/tree/master/examples/homekit_switch it works well.

2.https://github.com/espressif/esp-iot-solution
Cloning to "/ Users / exrector / esp-iot-solution / components / hmi / ugfx_gui / ugfx" ...
fatal: "https://git.ugfx.io/ugfx/ugfx.git/" not available: Failed to connect to git.ugfx.io port 443: Connection refused
fatal: failed to clone "https://git.ugfx.io/ugfx/ugfx.git" into a submodule under the path "/ Users / exrector / esp-iot-solution / components / hmi / ugfx_gui / ugfx"
Failed to clone 'components / hmi / ugfx_gui / ugfx' on second try, cancel

3.https://github.com/espressif/esp-iot-solution/tree/master/components/motor/servo/servo_motor
Too old project
CMake Error at CMakeLists.txt: 20 (register_component):
  Unknown CMake command "register_component".
CMake Warning (dev) in CMakeLists.txt:
  No cmake_minimum_required command is present. A line of code such as
    cmake_minimum_required (VERSION 3.18)
should be added at the top of the file. The version specified may be lower
  if you wish to support older CMake versions for this project. For more
  information run "cmake --help-policy CMP0000".
This warning is for project developers. Use -Wno-dev to suppress it.

4.https://github.com/espressif/esp-homekit-sdk/tree/master/examples/emulator
No way to check, we have a bug now https://github.com/espressif/esp-idf/issues/5962#issuecomment-706987249

5.https://github.com/maximkulkin/esp-homekit-demo/tree/master/examples/blinds
instruction https://github.com/maximkulkin/esp-homekit-demo/wiki/Build-instructions-ESP32
I would like to use this example, but I cannot put it together into a project.

exrector, 2021-06-01:
Hello @shahpiyushv 
Please add a working example of blinds

maxakuru, 2021-06-13:
@exrector I've just started working on a project for that, it's very early stages and not a minimal example, but I do have Homekit successfully reading/writing to a stepper motor connected to an ESP32 with this SDK (slightly modified)

Project is here: https://github.com/maxakuru/WBlinds
The motor and Homekit aren't coupled, but Homekit integration is [here](https://github.com/maxakuru/WBlinds/blob/main/src/homekit.cpp), and A4988 stepper driver controller is [here](https://github.com/maxakuru/WBlinds/blob/main/src/motor_a4988.cpp) (uses [FastAccelStepper](https://github.com/gin66/FastAccelStepper))

also, thanks for the SDK @shahpiyushv & espressif team!

arkn666l, 2023-05-25:
```
> @exrector I've just started working on a project for that, it's very early stages and not a minimal example, but I do have Homekit successfully reading/writing to a stepper motor connected to an ESP32 with this SDK (slightly modified)
> 
> Project is here: https://github.com/maxakuru/WBlinds The motor and Homekit aren't coupled, but Homekit integration is [here](https://github.com/maxakuru/WBlinds/blob/main/src/homekit.cpp), and A4988 stepper driver controller is [here](https://github.com/maxakuru/WBlinds/blob/main/src/motor_a4988.cpp) (uses [FastAccelStepper](https://github.com/gin66/FastAccelStepper))
> 
> also, thanks for the SDK @shahpiyushv & espressif team!
```

waiting for the integration all with homekit... there is no project working ok online :(

TheDevNino, 2023-08-25:
> ```
> ```
>
>
>
>
>
> waiting for the integration all with homekit... there is no project working ok online :(

is there anything new?

arkn666l, 2023-08-25:
only can make works with HAA from:

https://github.com/RavenSystem/