espressif/esp-homekit-sdk #10: Hardcoded Wi-Fi credentials -- does it work?

Cosmo, 2020-11-15:
Currently, my only option is to use the provisioning app and scan the QR code to get my device connected to my Wi-Fi.

For some reason, when using the hardcoded Wi-Fi data, the ESP32 tries to connect, then restarts, and tries again and again.
Same Wi-Fi for both scenarios.

I also created an additional 2.4 GHz Wi-Fi access point in my home to see if the problem exists. For SSID and password, I used simple characters (a-z, 0-9) to rule out any encoding issues or stuff like that.

Can someone please verify whether the hardcoded method works or point me in the right direction to debug my issue?

Thanks!

shahpiyushv, 2020-11-18:
@Cosmo , the primary difference in the Wi-Fi configuration when you use hard coded credentials is [this](https://github.com/espressif/esp-homekit-sdk/blob/040b0f301223ebc6995597328e5a5cc9f9739a02/examples/common/app_wifi/app_wifi.c#L273). Please try setting this value to false and let me know if it works.

Even if this indeed works, it would be nice if you can test the simple [Wi-Fi station example](https://github.com/espressif/esp-idf/tree/master/examples/wifi/getting_started/station) from esp-idf with your AP and if you see similar behaviour, file an issue on [esp-idf](https://github.com/espressif/esp-idf/issues). If you can get some sniffer captures, that would help too.

Cosmo, 2020-11-19:
@shahpiyushv Thank you very much! I wish I would have asked you earlier :)
I set `.capable` to `false` and now my ESP connects and stays connected — perfect!

So when I understand the flag correctly, PMF stands for protected management frames and either my access point doesn't support it or it's disabled.

After a quick check, my router shows that PMF is grayed-out because WPA2+WPA3 is enabled. WPA3 mandates the use of PMF, whereas in WPA2, it's optional. I guess I need to invest a bit more time into this.

Later, I tested the `WiFi station example`. I set my SSID and password via `menuconfig`, build and flashed on my device. It worked without any further changes.

So thanks again for saving me a lot of time and a headache. 
I really appreciate it. 👍

closed 2020-11-19 by Cosmo