espressif/esp-homekit-sdk #12: RF transmitter

lorenzofiamingo, 2020-11-26:
I'm trying to connect an RF transmitter to HomeKit.
The ideal would be to have a long press gesture button service, that sends the signal as long the user is holding the finger on the screen, but I didn't find anything in the apple-defined services (I tried the StatelessProgrammableSwitch, but I had to use it to handle other services). So I moved to a switch that waits 1s and then turns back to off. Despite my low experience I managed to create a working sample. The only problem is that if I repeatedly press the key in that second, the program buffers all the user requests creating an unwanted behavior (practically a blinking light). 
Is there a way to stop and restart incoming requests? Or a completely different but better way to handle this problem?
Sorry but I'm really new to this...

shahpiyushv, 2020-12-10:
@lorenzofiamingo , if the only issue you intend to solve right now is to prevent multiple requests getting accumulated, one option is to handle the first command that you receive, and return immediately. You operation can start in parallel on a different thread. If you receive another request before your work is done, just return `HAP_STATUS_RES_BUSY` as the status instead of `HAP_STATUS_SUCCESS`. You can check [this example](https://github.com/espressif/esp-homekit-sdk/blob/master/examples/fan/main/app_main.c#L196) to understand how you can return status.

Meanwhile, currently, are you blocking the write callback for 1 second for your operation to complete? Ideally, that should have been sufficient.

shahpiyushv, 2021-01-11:
Closing this issue as there hasn't been any further communication on it. Please feel free to reopen if required.

closed 2021-01-11 by shahpiyushv