espressif/esp-homekit-sdk #19: Which SDK should we use?

slompf18, 2020-12-31:
It seems espressif is developing two different sdks:

- [esp-apple-homekit-adk](https://github.com/espressif/esp-apple-homekit-adk)
- [esp-homekit-sdk](https://github.com/espressif/esp-homekit-sdk)

Which one should we use? Is the ADK just working for commercial projects with MFI?

shahpiyushv, 2020-12-31:
@slompf18, The esp-apple-homekit-adk is a port of Apple's ADK, whereas esp-homekit-sdk is completely home grown. You may use any SDK that you find suitable for your use case and convenient to use. Both of these are meant for hobbyists and both also have their MFi compliant counterparts if you want to build a commercials product.

slompf18, 2021-01-02:
Thanks for the answer. Can you give some guideline:
- Is performance the same?
- Are they both stable?
- Do they both support the same feature set? Are they feature complete compared to the spec?
- Do they both support IP and BLE?
- Will they both be maintained (in terms of the ADK updated if Apple updates)
- Which one may become part of the IDF repository?

timoschilling, 2021-01-02:
What’s the difference between this SDK and the MFi compliant one?

drewandre, 2021-01-04:
Wanted to add one more question to the list: which SDK is more similar to the MFi release from espressif? I worry that I'll get comfortable with the API from this SDK only to have it change drastically when requesting the MFi SDK.

shahpiyushv, 2021-01-04:
Here are the answers

> Is performance the same?

Performance should be similar as the underlying esp-idf is the same.

> Are they both stable?

The primary code base of both can be considered as stable, but with a minor difference. esp-homekit-sdk is developed in-house and works with all stable IDF releases 4.x. esp-apple-homekit-adk is a port of Apple's ADK, and it requires a custom mbedtls branch, which in itself is not part of any stable IDF release.

> Do they both support the same feature set? Are they feature complete compared to the spec?

Both are compliant with the HAP Spec R15

> Do they both support IP and BLE?

esp-homekit-sdk does not support BLE at all and may not support in future as well. Apple's ADK has support for BLE, but we haven't ported it in esp-apple-homekit-adk. The port could be added in future.

> Will they both be maintained (in terms of the ADK updated if Apple updates)

Yes. 

> Which one may become part of the IDF repository?

None of these will be part of esp-idf and will be maintained as separate projects.

> What’s the difference between this SDK and the MFi compliant one?

At API level it is the same, but in the MFi compliant variant, you can enable the MFi authentication options and Apple Wi-Fi Accessory Configuration (WAC provisioning).

> Which SDK is more similar to the MFi release from espressif? 

Both, esp-homekit-sdk and esp-apple-homekit-adk have their corresponding MFi variants. However, esp-homekit-sdk has been the primary MFi compliant SDK from Espressif since long.

slompf18, 2021-01-05:
Thanks for clarification. So I guess the ADK port is the better choice:

- Probably well tested by Apple, as it runs on many devices.
- Less effort as new features will be implemented by Apple and have "only" to be ported.
- May support BLE in the future.

The only disadvantages I see are:

- It needs a special version of IDF. But it seems, like this will be solved in the future: [link](https://github.com/espressif/esp-idf/issues/4958).
- It does not work for ESP8266. But ESP8266 does not work for commercial products anyway.

closed 2021-01-05 by slompf18

drewandre, 2021-01-05:
@shahpiyushv thanks for answering everything!

I have one last question regarding MFi authorization. I assumed it would be s/w auth, but I see in the MFi menuconfig that you can configure I2C pins for what I assume to be h/w authorization with a MFi co-processor. Is this the case?

If so, could you briefly describe the process for obtaining a co-processor? Does espressif provide a co-processor distributor? We've purchased MFi co-processors from Avnet before but just want to know if we need to start planning for this in our board design. Thanks.

shahpiyushv, 2021-01-05:
@slompf18 some clarifications

> Probably well tested by Apple, as it runs on many devices.

This could be true, but I am not 100% sure about this as almost all our customers have been using esp-homekit-sdk as it has been around for a longer time. Even other chip vendors have had their own custom SDKs and the ADK came later.

> Less effort as new features will be implemented by Apple and have "only" to be ported.

The efforts here would be more on Espressif's side and not end users.

shahpiyushv, 2021-01-05:
@drewandre you may choose to use either SW auth or HW auth, both are supported by the SDK. MFi co-processor can be procured from the MFi portal from the distributors authorised by Apple, but of course, it isn't required if you choose the SW token authentication.

slompf18, 2021-01-05:
> This could be true, but I am not 100% sure about this as almost all our customers have been using esp-homekit-sdk as it has been around for a longer time. Even other chip vendors have had their own custom SDKs and the ADK came later.
 
 I was thinking on other devices, like smart TVs, etc. There it is probably used, too.

shahpiyushv, 2021-01-14:
@slompf18 

> I was thinking on other devices, like smart TVs, etc. There it is probably used, too.

This is unlikely to be the case since HomeKit wasn't meant for such types of devices.

dguerri, 2021-01-14:
@shahpiyushv sorry for the direct question: should you start a new project today, which one you would go for and why :)

slompf18, 2021-01-14:
> @slompf18
> 
> > I was thinking on other devices, like smart TVs, etc. There it is probably used, too.
> 
> This is unlikely to be the case since HomeKit wasn't meant for such types of devices.

Some new Sony and LG smart TVs are accessible via HomeKit. Good chance they are using the ADK of Apple.

shahpiyushv, 2021-01-15:
> @shahpiyushv sorry for the direct question: should you start a new project today, which one you would go for and why :)

@dguerri  I may be biased because most of the code in esp-homekit-sdk is written by me, whereas esp-apple-homekit-adk is just a port of Apple's ADK, the code for which I myself do not know much :)

The APIs in both are totally different and you can check for yourself which ones you prefer. One advantage with esp-homekit-sdk is that it uses most stuff from esp-idf (like crypto, webserver, etc.) and has integrations like aws-iot, ESP RainMaker, Unified Provisioning etc. ADK on the other hand may support some additional profiles which may not be part of esp-homekit-sdk and it has also has support for BLE, which may be ported into esp-apple-homekit-adk sometime in future.

yemirabalaraju, 2021-08-09:
Does the Demo sdk can be used for SW authentication without getting Mfi variant esp32  SDK/ADK..?

If yes, which part of the code has to look for for SW authentication?

shahpiyushv, 2021-08-09:
@yemirabalaraju , HomeKit software authentication requires the SW Tokens from Apple which only MFi Licensees are eligible to get. Moreover, the specification for that are also open only for MFi Licensees. So, to try that out, you would have to use the MFi variant of the SDK/ADK.

yemirabalaraju, 2021-09-27:
Hi, 
a generic question.. 
I have option for going by Software token Auth. or MFi hardware chip based Auth.  Which approach is best, ? 
Which one is more secure ? HW or SW base auth..
What is industry trend in method of Authentication process..

shahpiyushv, 2021-09-27:
SW token auth may be cheaper since you save in the MFi chip. The pair setup is slightly slower though, but it should not matter much since it comes into picture just once when the accessory is being configured. Apple would be in a batter position to suggest which one to use.