espressif/esp-homekit-sdk #21: DHT22 example

maxiiiiiiii, 2021-01-15:
I would like to build an Apple Homekit weather station with the DHT22 temperature and humidity sensor.
I am not very familiar with coding yet and wanted to ask if it was possible adding a little program in the example-section of  the esp-homekit-sdk that pulls the temperature and humidity data from the DHT22 and provides it to Homekit.

In combination with esp32-deepsleep and some lithium ion batteries this would result in a very decent and economic Homekit weather station.

Thx a lot
maxi

timoschilling, 2021-01-15:
I think deepsleep and this SDK aren't compatible. If your esp goes into deepsleep the sensor will be shown as unavailable in the Home app. For that what you want to do you need BLE, but BLE isn't implemented at the moment.

Hint: Build a Hub for example with homebridge or something like that which stores the sensor values of the esp and is always available to HomeKit.

AramVartanyan, 2021-01-15:
@maxiiiiiiii, this example could be helpful to you as a starting point:
[Watering + DHT22 on Battery and Solar](https://github.com/AramVartanyan/esp-homekit-sdk/tree/master/examples/waterbatt)

AramVartanyan, 2021-01-15:
Actually it is very compatible with esp32.
There is just slight difference in the GPIO initialization and the GPIO range in Kconfig.
Everything else is originally done for esp32, including the DHT22.
I have test it on both platforms.

maxiiiiiiii, 2021-01-20:
> I think deepsleep and this SDK aren't compatible. If your esp goes into deepsleep the sensor will be shown as unavailable in the Home app. For that what you want to do you need BLE, but BLE isn't implemented at the moment.

Thank you. After long hours of trying and comparing code i finally was able to build a homekit temperature and humidity sensor.

As timoschilling already anticipated, deepsleep is not an option for a homekit sensor, because the Wi-Fi has to be turned on all the time.
My idea now: I will use one esp32 as a kind of bridge that is always turned on and holds up all the sensor-data for eventual homekit-requests. It runs the homkit-sensor program I just finished, only the sensor-read gets sourced out. 
A second esp32, equipped with a DHT22-sensor-unit will provide its data over ESP-NOW every now and then and run on lithium-battery.

Unluckily the only ESP-NOW example provided by espressif in the example section of its ESP-IDF-github (link below), is overcomplicated and hard to understand. I only need to send on one esp and receive on the other esp. All the other ESP-NOW examples I found on Google were written for Arduino.

https://github.com/espressif/esp-idf/tree/master/examples/wifi/espnow

mbuckaway, 2021-01-21:
@maxiiiiiiii:

Here is an example I created to read the DH22 and connect to Homekit:[ ESP32-homekit-temp](https://github.com/mbuckaway/esp32-homekit-temp). It also supports reading the battery level on supported boards. The code supports the ESP32 IDF. The code is mostly complete and I had it working on a device with a battery. Of course, the 1800ma/h battery only lasted about two days.

Be careful implementing a weather station for homekit. Some of the limits attached to "standard" homekit devices are outside of normal sensor ranges, which will cause the device to fail to connect to the Home App. For example, the CURRENT_TEMPERATURE device has a limit from 0 to 100C and requires a custom device (or custom implementation with different limits) to go beyond this range.

My solution to the battery problem (I wanted a homekit outside temperature sensor) was to setup a modbus temperature sensor outside, and run a wire to the ESP32 inside which is plugged into a USB power supply. At some point, the battery would have to be charged anyways, and this eliminates that need. There are lots of modbus weather sensors on Aliexpress anyways for wind, temperature, etc. so one ESP32 can run everything.