espressif/esp-homekit-sdk #22: Commercial use

Jaarson, 2021-01-21:
Hi,

I have a question about commercial use. Would the following cases be treated as a commercial use?

1. Video course for developers/makers that includes building a project based on SDK. Then selling this course.
2. Sell a kit for self-assembly, for developers/makers of some kind of device e.g. a smart light

shahpiyushv, 2021-01-22:
@Jaarson , I don't think this qualifies as a commercial use case. However, note that I have no authority to decide what is commercial and what is not, as that is Apple's prerogative .

Jaarson, 2021-01-23:
Thank you @shahpiyushv ! So as I understood, in a nutshell, Espressif is following (or inherits) Apple's commercial use restrictions, and doesn't add its own/additional restrictions?

shahpiyushv, 2021-01-25:
@Jaarson , that's right.

Jaarson, 2021-01-25:
Thanks! All clear, I'm closing this.

closed 2021-01-25 by Jaarson