espressif/esp-homekit-sdk #23: Controlling multiple devices with one esp32

manvish, 2021-01-26:
Hello Developers,
I have successfully implemented controlling of single light bulb from Accessory button in Home App.
Now I want to add more things like PWM generation on 3 channels and 3 more GPIO to control from different buttons on APP.
Is there any example similar to my requirement. ?

shahpiyushv, 2021-01-27:
@manvish , you have 2 options

1) Add an additional lightbulb service to the same accessory by repeating the lines from [here](https://github.com/espressif/esp-homekit-sdk/blob/389189abd7c1965d70eb3ddc7a19a8f0313f1fc8/examples/lightbulb/main/app_main.c#L183).
2) Define your primary accessory as a bridge and then add some lightbulb accessories to it, as per [this example](https://github.com/espressif/esp-homekit-sdk/blob/master/examples/bridge/main/app_main.c)

The actual hardware interfacing code will have to be written by you as it is very use case specific.

manvish, 2021-01-30:
Thanks.
Can you tell me how to add three different slider on Home App, to change pwm on 3 different pins

shahpiyushv, 2021-02-01:
@manvish are the 3 different pins for 3 different lights or for RGB?

If it is for 3 different lights, please create 3 different lightbulb services and map brightness characteristic with your pwm duty cycle.

If it is for RGB, you will have to convert RGB to HSV (hue, saturation, value) as HomeKit supports HSV format for colour lights. You can find a conversion function [here](https://github.com/espressif/esp-homekit-sdk/blob/389189abd7c1965d70eb3ddc7a19a8f0313f1fc8/examples/lightbulb/main/lightbulb.c#L79).

shahpiyushv, 2021-03-04:
Closing this issue since it has been answered.

closed 2021-03-04 by shahpiyushv