espressif/esp-homekit-sdk #24: Sync service state manually when using physical button

cocofly59, 2021-02-23:
Hi,

I'm making a basic connected light with a physical switch using the lightbulb example. Everything works perfectly except that when  I'm using the physical switch, the light state is not sync with the server (for example, the iPhone shows the light is on instead of off).

I'm unable to find the way to notify the server that the light' state has changed. Any idea?

shahpiyushv, 2021-02-23:
@cocofly59 , to report values to the iPhone via async event notifications, you just need to use this API: [`int hap_char_update_val(hap_char_t *hc, hap_val_t *val);](https://github.com/espressif/esp-homekit-sdk/blob/389189abd7c1965d70eb3ddc7a19a8f0313f1fc8/components/homekit/esp_hap_core/include/hap.h#L1028) 

You can find sample usage [here](https://github.com/espressif/esp-homekit-sdk/blob/389189abd7c1965d70eb3ddc7a19a8f0313f1fc8/examples/smart_outlet/main/app_main.c#L229). I hope this answers your query.

cocofly59, 2021-02-23:
Thank you, it mostly answer my query.

I will add that you need to use the function '_hap_char_t *hap_serv_get_char_by_uuid(hap_serv_t *hs, const char *uuid)_ ' to access the characteristic you want to change (and '_const hap_val_t *hap_char_get_val(hap_char_t *hc)_' if you want to get the old value.

closed 2021-02-23 by cocofly59