espressif/esp-homekit-sdk #26: Adaptive Lighting

mgoeller, 2021-03-04:
Does the commercial version of this SDK have an example and support Adaptive Lighting for Lightbulbs for Colour Temperature Characteristic?

shahpiyushv, 2021-03-05:
@mgoeller , not yet, but will be added soon.

shahpiyushv, 2022-04-12:
@mgoeller , we do support this in the MFI compliant esp-homekit-adk. We can give you the access if you raise a request for it on the commercial SDK support account.

mgoeller, 2022-04-12:
I am using the official glab.espressif.com HomeKit SDK, there is no possibility to raise a request? Or which one are you talking about? Do you have an email address I can contact?

mgoeller, 2022-09-09:
I did this, but unfortunately we do not have access yet to neither HW nor SW authentication. So how do we provision the ESP without using WAC/2?

You official esp-homekit-sdk still does not support Adaptive Lighting, the new Lock Access service etc.