espressif/esp-homekit-sdk #28: Document concurrency model

lukehoersten, 2021-03-07:
I'm looking for more information on the concurrency model of the ESP HomeKit SDK HAP implementation. Does the HAP Core run it's own thread and then the App gets its own thread, etc? Thread safety considerations would be helpful as well.

shahpiyushv, 2021-03-10:
Since entire HomeKit communication happens over HTTP, all the read/write callbacks are invoked in the context of the [esp_http_server](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/esp_http_server.html). The various events (like HAP_EVENT_CTRL_PAIRED) are invoked using the [esp_event_loop](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/esp_event.html). Whenever application calls `hap_char_update_val()`, the value change event is queued and notification sent to controllers, again in the esp_http_server task's context. There is also another hap-loop task for some internal operations and state machine.

The `app_main()` function itself is invoked in the context of the main task, about which, you can read [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/general-notes.html#application-startup).

Ideally, the `hap_char_update_val()` API would need some protection, since it can get invoked from multiple task contexts, but currently it is missing. Other APIs can also have protection, but a lot of those aren't really required to be invoked from different tasks .

I hope this answers your question.

lukehoersten, 2021-03-12:
Yes this is super helpful - thank you.

Is it safe for me to call `hap_` functions from multiple tasks?