espressif/esp-homekit-sdk #30: bridge mode does not work  with more accessories nodes,

MrutyunjayPatel, 2021-03-11:
hi,
We were evaluating homekit sdk for bridged mode accessories .
using the bridge example with fan devices, extended the example to check if it can accommodate 32,64,128 Devices.

**using IDF version 4.2.0**

```
#define NUM_BRIDGED_ACCESSORIES 64
```
also we added new characteristic  of  speed to the fan example 

```
#ifdef (FAN_WITH_CUSTOM_SPEED25)
        hap_char_t *rotation_speed = hap_char_rotation_speed_create(50.0);
        hap_char_float_set_constraints(rotation_speed, 0, 100, 25.0);
        hap_serv_add_char(service, rotation_speed);
#else
	#ifdef (FAN_WITH_CUSTOM_SPEED10)
        hap_char_t *rotation_speed = hap_char_rotation_speed_create(50.0);
        hap_char_float_set_constraints(rotation_speed, 0, 100, 10.0);
        hap_serv_add_char(service, rotation_speed);
	#endif
#endif
```

**During adding accessories (Bridge) to ipad home app the session gets terminated prematurely.**

**----------Fails to add 32/64 device in Home-----**
```
W ( 5400) httpd_parse: parse_block: request URI/header too long
W ( 5401) httpd_txrx: httpd_resp_send_err: 414 URI Too Long - URI is too long for server to interpret
HomeKit Session terminated
```
**Needs to increase the http url length to accommodate more devices  **
------http 1536 ----Ok--------
```
#
# HTTP Server
#
CONFIG_HTTPD_MAX_REQ_HDR_LEN=512
CONFIG_HTTPD_MAX_URI_LEN=1536
CONFIG_HTTPD_ERR_RESP_NO_DELAY=y
CONFIG_HTTPD_PURGE_BUF_LEN=32
# CONFIG_HTTPD_LOG_PURGE_DATA is not set
# CONFIG_HTTPD_WS_SUPPORT is not set
# end of HTTP Server
```
we could add 64 first then add another 64 Total 128. 
but could not set or get the read the status of all 128.

added a bulk on and off in home kit app as **Home Away(all off)** and **Enter Home(all on)**     

**at 128 device count** 
```
Allocated buffer of size 11746 for the large PUT
Failed to parse HTTPD JSON Data
Freed allocated buffer for PUT
Decryption error/Connection lost. Marking session as invalid
HomeKit Session terminated
``` 

Tried to increase CONFIG_HTTPD_MAX_URI_LEN=4096

Then got bored to add the 128 device again in the homekit app had to click and assign each fan to a room :( 

Thanks and Regards

shahpiyushv, 2021-03-12:
@MrutyunjayPatel  to see the actual http data being sent and received by the HAP core, you may call `hap_http_debug_enable()` before `hap_start()`. It will help understand if the HAP core is indeed getting correct from the http server module or not.

It is quite possible that there wasn't enough heap memory to parse the received JSON. Check the APIs [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/heap_debug.html#heap-information) to do some heap analysis at your end. If you indeed require 100+ accessories, I would suggest that you use the ESP32 Wrover module which has external RAM.

Meanwhile, since you are getting issues only while getting/setting all values, and not during pairing, is it really required to remove and add back the bridge, going through the cumbersome process? Moreover, what happens if you just add a single bridge during HomeKit pairing and in the next boot cycle, add all the remaining bridged accessories?

MrutyunjayPatel, 2021-03-15:
@shahpiyushv .
yes the error was due to http phrasing limitations,
As i mentioned, I had error only during adding 128 device in a single go. no issues if adding in batch of 16/32/64.
yes the memory does get used up quickly as we go on adding more accessories.

**For future reference to to others , Memory usage for BRIDGE MODE with 64 and 128 devices .** 
CONFIG_HTTPD_MAX_REQ_HDR_LEN=512
CONFIG_HTTPD_MAX_URI_LEN=1536             increase this if needed be.
--------------------
**64 fans** 
--------------------
I (57587) HAP Bridge: Free Memory Available' is 137128 bytes
I (57587) HAP Bridge: MinEver Free Memory Available' is 95436 bytes
-------------------------------------------------------------------------------
**128 fans (64+64).**
-------------------
I (25657) HAP Bridge: Free Memory Available' is 59692 bytes
I (25657) HAP Bridge: MinEver Free Memory Available' is 42652 bytes
-------------------------------------------------------------------------------
After some bulk operations all-on/all-off
I (2451487) HAP Bridge: Free Memory Available' is 59408 bytes
I (2451487) HAP Bridge: MinEver Free Memory Available' is 5580 bytes

I don't have ESP32 Wrover module , to further evaluate this.

Thanks and regards,
 Mrutyunjay

Have a nice day !

closed 2021-03-15 by MrutyunjayPatel