espressif/esp-homekit-sdk #36: Examples not work!

lewisxhe, 2021-04-07:
I cannot add devices using Apple Home or RainMaker. 

After scanning the QR code, the serial terminal displays
```
W (29853) wifi:Error! Should use default active scan time parameter for WiFi scan when Bluetooth is enabled!!!!!!
``` 

Below is the complete log .

```
rst:0x1 (POWERON_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7148
load:0x40078000,len:13212
ho 0 tail 12 room 4
load:0x40080400,len:4568
0x40080400: _init at ??:?

entry 0x400806f4
I (31) boot: ESP-IDF v4.2-408-gc9cf7bcb0-dirty 2nd stage bootloader
I (31) boot: compile time 16:42:44
I (31) boot: chip revision: 1
I (35) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (44) boot.esp32: SPI Speed      : 40MHz
I (47) boot.esp32: SPI Mode       : DIO
I (52) boot.esp32: SPI Flash Size : 4MB
I (56) boot: Enabling RNG early entropy source...
I (62) boot: Partition Table:
I (65) boot: ## Label            Usage          Type ST Offset   Length
I (72) boot:  0 sec_cert         unknown          3f 00 0000d000 00003000
I (80) boot:  1 nvs              WiFi data        01 02 00010000 00006000
I (87) boot:  2 otadata          OTA data         01 00 00016000 00002000
I (95) boot:  3 phy_init         RF data          01 01 00018000 00001000
I (102) boot:  4 ota_0            OTA app          00 10 00020000 00190000
I (110) boot:  5 ota_1            OTA app          00 11 001b0000 00190000
I (117) boot:  6 factory_nvs      WiFi data        01 02 00340000 00006000
I (125) boot:  7 nvs_keys         NVS keys         01 04 00346000 00001000
I (132) boot: End of partition table
I (137) boot_comm: chip revision: 1, min. application chip revision: 0
I (144) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x44284 (279172) map
I (259) esp_image: segment 1: paddr=0x000642ac vaddr=0x3ffbdb60 size=0x03fc0 ( 16320) load
I (266) esp_image: segment 2: paddr=0x00068274 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at /mnt/d/Desktop/wled/esp-idf-4.2/components/freertos/xtensa/xtensa_vectors.S:1730

I (267) esp_image: segment 3: paddr=0x00068680 vaddr=0x40080404 size=0x07998 ( 31128) load
I (289) esp_image: segment 4: paddr=0x00070020 vaddr=0x400d0020 size=0x1121f0 (1122800) map
0x400d0020: _stext at ??:?

I (716) esp_image: segment 5: paddr=0x00182218 vaddr=0x40087d9c size=0x14f50 ( 85840) load
0x40087d9c: r_lld_evt_schedule at ??:?

I (770) boot: Loaded app from partition at offset 0x20000
I (770) boot: Disabling RNG early entropy source...
I (770) cpu_start: Pro cpu up.
I (774) cpu_start: Application information:
I (779) cpu_start: Project name:     lightbulb
I (784) cpu_start: App version:      b3ca71e-dirty
I (789) cpu_start: Compile time:     Apr  7 2021 16:42:25
I (795) cpu_start: ELF file SHA256:  32b8d148af84046c...
I (801) cpu_start: ESP-IDF:          v4.2-408-gc9cf7bcb0-dirty
I (808) cpu_start: Single core mode
I (812) heap_init: Initializing. RAM available for dynamic allocation:
I (819) heap_init: At 3FF80000 len 00002000 (8 KiB): RTCRAM
I (825) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (831) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (838) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (844) heap_init: At 3FFCDEA0 len 00012160 (72 KiB): DRAM
I (850) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM
I (856) heap_init: At 40078000 len 00008000 (32 KiB): IRAM
I (862) heap_init: At 4009CCEC len 00003314 (12 KiB): IRAM
I (869) cpu_start: Pro cpu start user code
I (886) spi_flash: detected chip: gd
I (887) spi_flash: flash io: dio
I (887) cpu_start: Starting scheduler on PRO CPU.
Keystore initialised
Accessory is not Paired with any controller
Database initialised. Accessory Device ID: 07:DC:1B:58:74:85
HAP Initialization succeeded. Version : 4.0-b3ca71e
I (933) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 
I (943) app_hap_setup_payload: -----QR Code for HomeKit-----
I (943) app_hap_setup_payload: Scan this QR code from the Home app on iOS

  █▀▀▀▀▀█   █▀  █▀▀▀▀▀█
  █ ███ █ █▀▄▀▀ █ ███ █
  █ ▀▀▀ █ ██▀ █ █ ▀▀▀ █
  ▀▀▀▀▀▀▀ █ █ ▀ ▀▀▀▀▀▀▀
  ▀ ▀███▀ ▄▀ ▄█▄██▀██ ▄
  █▄█▄▀▄▀▀▀█▀▄█  ▀  ▄█▄
    ▀▀  ▀▀▄▄▀ ▀▄▀▄ ▄▄█▀
  █▀▀▀▀▀█ ▄  ▄█▀▄█ █ ▀▄
  █ ███ █ █▀█  ▄▀▀█▀  ▀
  █ ▀▀▀ █ ▀█▄ █▄ ▄ █▀ ▄
  ▀▀▀▀▀▀▀ ▀▀ ▀▀▀  ▀▀

I (1033) app_hap_setup_payload: If QR code is not visible, copy paste the below URL in a browser.
https://espressif.github.io/esp-homekit-sdk/qrcode.html?data=X-HM://00527813XES32
MFi auth not supported. Falling back to HAP_MFI_AUTH_NONE
I (1063) wifi:wifi driver task: 3ffd0f58, prio:23, stack:6656, core=0
I (1063) system_api: Base MAC address is not set
I (1063) system_api: read default base MAC address from EFUSE
I (1083) wifi:wifi firmware version: 3c8d1dd
I (1083) wifi:wifi certification version: v7.0
I (1083) wifi:config NVS flash: enabled
I (1083) wifi:config nano formating: disabled
I (1083) wifi:Init data frame dynamic rx buffer num: 32
I (1093) wifi:Init management frame dynamic rx buffer num: 32
I (1093) wifi:Init management short buffer num: 32
I (1103) wifi:Init dynamic tx buffer num: 32
I (1103) wifi:Init static rx buffer size: 1600
I (1113) wifi:Init static rx buffer num: 10
I (1113) wifi:Init dynamic rx buffer num: 32
I (1113) wifi_init: rx ba win: 6
I (1123) wifi_init: tcpip mbox: 32
I (1123) wifi_init: udp mbox: 10
I (1133) wifi_init: tcp mbox: 6
I (1133) wifi_init: tcp tx win: 5744
I (1133) wifi_init: tcp rx win: 5744
I (1143) wifi_init: tcp mss: 1440
I (1143) wifi_init: WiFi IRAM OP enabled
I (1153) wifi_init: WiFi RX IRAM OP enabled
Setup ID: ES32
HAP Main Loop Started
mDNS initialised
Registering HomeKit web handlers
Announcing _hap._tcp mDNS service
I (1173) wifi_prov_scheme_ble: BT memory released
I (1173) app_wifi: Starting provisioning
I (1183) phy_init: phy_version 4660,0162888,Dec 23 2020
I (1273) wifi:mode : sta (30:ae:a4:49:93:00)
I (1273) BTDM_INIT: BT controller compile version [cbc69c8]
I (1283) phy_init: phy_version 4660,0162888,Dec 23 2020
I (1633) wifi_prov_mgr: Provisioning started with service name : PROV_499300 
I (1633) app_wifi: Provisioning started
I (1643) app_wifi: -----QR Code for ESP Provisioning-----
I (1643) app_wifi: Scan this QR code from the phone app for Provisioning.

  █▀▀▀▀▀█ ▀▀▀█▄█    ▄▄█   ▀ █▀▀▀▀▀█
  █ ███ █  ▀▄█ █▄ ▀▄▄▀▀█▄▄▀ █ ███ █
  █ ▀▀▀ █  ▄▀█▀▄▀ ▀▄▄▀▀██▄█ █ ▀▀▀ █
  ▀▀▀▀▀▀▀ █▄▀ █▄█▄█ ▀▄▀ █ ▀ ▀▀▀▀▀▀▀
  ▄ ▀ █▄▀▄▄▄▄▀▄▄█▄▀▀█▄█▄█▀▀█ ▀▄ ▄▀
  █ ▀ ▀▄▀█▄███▄  ▀█▄█▄▀▀█▀█ ▄█ ▀▄▄█
  ▀▀▀▀▀ ▀ █ ▀▀▀▄▄██▄█▀█ ▀██▀▀▀█▄▄▀
  █▀▀▀▀▀█  █▄█▀▀▀██ ▄▀▄ █▄█ ▀ █ ▄ ▄
  █ ███ █ ███▄█▀▀█▀▄█▄▄ ▀██▀▀▀▀▄▄▀▀
  █ ▀▀▀ █ ▄██▀ ▄█▀█ █▀ ▀▀███▄▀█ █▄█
  ▀▀▀▀▀▀▀ ▀ ▀▀  ▀▀ ▀     ▀▀▀▀▀▀

I (1853) app_wifi: If QR code is not visible, copy paste the below URL in a browser.
https://espressif.github.io/esp-jumpstart/qrcode.html?data={"ver":"v1","name":"PROV_499300","pop":"a4499300","transport":"ble"}ansport":"ble"}
I (1873) app_wifi: Provisioning Started. Name : PROV_499300, POP : a4499300                                     !
W (29853) wifi:Error! Should use default active scan time parameter for WiFi scan when Bluetooth is enabled!!!!!!
```

shahpiyushv, 2021-04-07:
@lewisxhe , scanning the smaller QR code from Home app will work only after the accessory is provisioned. The HomeKit based provisioning (called Apple WAC Provisioning) which works from the Home app is supported only in the MFi variant of the SDK.

Please use the ESP Provisioning phone apps from [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/provisioning/provisioning.html#provisioning-tools). You will need the ESP BLE Prov app since you are using BLE provisioning here. The message `W (29853) wifi:Error! Should use default active scan time parameter for WiFi scan when Bluetooth is enabled!!!!!!` can be ignored since that is just some warning printed from the Wi-Fi library but it does not affect any functionality.

lewisxhe, 2021-04-08:
Hi, @shahpiyushv 
Sorry, I saw the wrong configuration.
Now that I use ESP Provisioning phone apps to configure the device, what app should I use to control the device? Can Android devices be controlled? Regarding HomeKit, I will configure it temporarily.

shahpiyushv, 2021-04-08:
@lewisxhe , the esp-homekit-sdk in particular provides only Provisioning facility for Android. However, if you want full control from Android, you can check out our other product, [ESP RainMaker](https://github.com/espressif/esp-rainmaker). It also has a [HomeKit + RainMaker combo example](https://github.com/espressif/esp-rainmaker/tree/master/examples/homekit_switch).

lewisxhe, 2021-04-08:
Okay, thank you for your answer.

closed 2021-04-08 by lewisxhe