espressif/esp-homekit-sdk #37: Please add Sprinklers, Faucets, Shower Systems to Category Identifiers

joeyhagedorn, 2021-04-07:
As is available in the Apple HomeKit ADK, and recent versions of the HomeKit specification, please add the Sprinklers, Faucets, and Shower Systems CIDs for accessories of this type.

See: https://github.com/apple/HomeKitADK/blob/4967f698bdcf0af122e13e986a2c9b595a68cdc5/HAP/HAP.h

`--- a/components/homekit/esp_hap_core/include/hap.h
+++ b/components/homekit/esp_hap_core/include/hap.h
@@ -222,6 +222,9 @@ typedef enum {
     HAP_CID_AIR_CONDITIONER,
     HAP_CID_HUMIDIFIER,
     HAP_CID_DEHUMIDIFIER,
+    HAP_CID_SPRINKLERS = 28,
+    HAP_CID_FAUCETS = 29,
+    HAP_CID_SHOWERSYSTEMS = 30,
     HAP_CID_MAX,
 } hap_cid_t;`

jimmyasyraf, 2022-12-30:
This can be closed. Recently added in #98

shahpiyushv, 2023-01-03:
@joeyhagedorn , sorry I missed this PR earlier.

closed 2023-01-03 by shahpiyushv