espressif/esp-homekit-sdk #38: SDK multiple sensor support?

Lebowski, 2021-04-12:
I was wondering, does the SDK support having multiple sensors on one accessory? I'm trying to use multiple DS18B20s on one accessory. I've only been able to get 1 to work. Can't figure this out.

shahpiyushv, 2021-04-12:
@Lebowski , can you elaborate on where exactly are you stuck? If you have the hardware part working for the 2 sensors, you can create 2 different (but similar) services for each, under the same accessory

Lebowski, 2021-04-12:
So I've adapated the fan example to instead be a temperature sensor (using a DS18B20). I can wire up another sensor in a similar fashion using a different GPIO pin however I can't figure out how to actually use that 2nd sensor.

timoschilling, 2021-04-12:
HomeKit has the limitation that one accessory can only have each characteristic once. So your plan isn't possible in this way. But you can create a brigade accessory, so your esp can provide one accessory for each of your temp sensors.

Lebowski, 2021-04-12:
So what I'd have to do is make my device a bridge accessory and then have multiple sensors on it or would I need a bridge accessory and multiple devices each with 1 sensor?

timoschilling, 2021-04-12:
> make my device a bridge accessory and then have multiple sensors on it

Lebowski, 2021-04-12:
> > make my device a bridge accessory and then have multiple sensors on it
> 
> one bridge accessory can have multiple accessories, each of this accessories will represent one sensor.

Awesome. Thank you so much for this info.

Lebowski, 2021-04-12:
I've managed to get it to sort of work. My issue now lies with specifying which GPIO port to use depending on the accessory that's being accessed (temp sensor 1, 2, and 3) The way I'm currently doing it is comparing (char *)serv_priv to the desired name (Sens-0, Sens-1, Sens-2) and then reading the appropriate GPIO port. A few issues I'm encountering is that for some reason it's assigning the ports to the wrong accessories and when 1 sensor is unplugged it just copies the data from the previous read instead of returning HAP_FAIL

closed 2021-04-12 by Lebowski