espressif/esp-homekit-sdk #40: Lightbulb example not pairing on ESP8266

twlawrenceko, 2021-04-29:
Hello all,

I have tried to flash lightbulb example onto esp-12f variant but pairing with Home app failed with the message `Couldn't add Esp-Light'. Below is the following serial monitor that showed during the pairing process.

```
######## Starting Pair Setup ########
Pair Setup M1 Received
Pair Setup M2 Successful
Pair Setup M3 Received
Using pair-setup without MFi.
Pair Setup M4 Successful
Pair Setup M5 Received
Pair Setup Successful for FB660B62-3EF5-4F48-9848-4321FDF7B8D5
Updated state number to 2
Re-announcing _hap._tcp mDNS service
Cleaning Pair Setup Context
Cleaning Pair Setup Context
######## Starting Pair Verify ########
Pair Verify M1 Received
Pair Verify M2 Successful
Pair Verify M3 Received
HomeKit Session active
Pair Verify Successful for FB660B62-3EF5-4F48-9848-4321FDF7B8D5
Decryption error/Connection lost. Marking session as invalid
HomeKit Session terminated
######## Starting Pair Verify ########
Pair Verify M1 Received
Pair Verify M2 Successful
```

I have got smart outlet example working, it seems like it somehow failed during the decryption error state. 

Thank you.

shahpiyushv, 2021-04-29:
@twlawrenceko , your issue may be similar to #4 , and the solution can be found here: https://github.com/espressif/esp-homekit-sdk/issues/4#issuecomment-712879446.

Basically, please search for the below config option in your sdkconfig file and change it from y to n.

```
CONFIG_NEWLIB_NANO_FORMAT=y
```

Let me know if it fixes the issue for you.

twlawrenceko, 2021-04-29:
@shahpiyushv , Thank you, that solved the issue.

twlawrenceko, 2021-04-29:
Once I click any button of the esp-light under the Home app, the esp8266 reboots itself, is this normal?

shahpiyushv, 2021-04-29:
@twlawrenceko , the lightbulb example was primarily built for ESP32-Wrover-Kit and the driver part may not work as is on ESP8266, because of difference in the timer/ledc modules. You may have to refer the [ledc example under ESP8266-RTOS-SDK](https://github.com/espressif/ESP8266_RTOS_SDK/tree/master/examples/peripherals/ledc) and change the lightbulb.c file accordingly.

closed 2021-04-30 by twlawrenceko