espressif/esp-homekit-sdk #41: Hardcode WiFi not able to change after first setup

twlawrenceko, 2021-05-05:
Hello,

It seems like after the first set of a particular AP name and password, even if I change to a new AP name and password inside `make menuconfig` it does not connect to the new AP, instead, I am getting a result of `app_wifi: Disconnected. Connecting to the AP again...`

I have also tried to include WiFi log info under `make menuconfig` but it does not seem to print those out in the serial monitor.

Is there any reason for this?

P.s. I am using smart outlet and lightbulb example from the SDK.

Thanks

twlawrenceko, 2021-05-05:
It turns out if I copy the examples directory into a new directory and use `make menuconfig` command, it will not affect the ESP's WiFi bootloader section, why is this and how can I solve this issue?

shahpiyushv, 2021-05-07:
@twlawrenceko , do you mean that you see the problem only when you move the example folder somewhere else? Using the default folder structure and examples, I verified that the network indeed changes if I change it via menuconfig. Have you checked if the network name reflects correctly in your sdkconfig file? `CONFIG_APP_WIFI_SSID` and `CONFIG_APP_WIFI_PASSWORD` are the relevant options to check. 

The second question isn't really clear. Did you copy the entire examples folder or just the smart_outlet/lightbulb folder? Did you adjust the paths in the example Makefile so that it can find the appropriate components?

twlawrenceko, 2021-08-22:
> @twlawrenceko , do you mean that you see the problem only when you move the example folder somewhere else? Using the default folder structure and examples, I verified that the network indeed changes [...]

Hello, for the second question, I mean of creating another directory which is under the same parent as `examples` folder. Since the new folder that I created `projects` is at the same level of `examples` folder, shouldn't any of the below path be changed?

├───esp-homekit-sdk
│   ├───examples
│   │    smart_outlet
│   └───projects
│   │    smart_outlet

```
HOMEKIT_PATH ?= $(abspath $(shell pwd)/../..)
COMMON_COMPONENT_PATH ?= $(abspath $(shell pwd)/../common)

PROJECT_NAME := smart_outlet
EXTRA_COMPONENT_DIRS += $(HOMEKIT_PATH)/components/
EXTRA_COMPONENT_DIRS += $(HOMEKIT_PATH)/components/homekit
EXTRA_COMPONENT_DIRS += $(COMMON_COMPONENT_PATH)

include $(IDF_PATH)/make/project.mk
```

How should I adjust the path accordingly with my directory setup?

Below is the error log:
```/user/esp/esp-idf=IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.4-dev-1849-g8e3e65a47\" -DESP_PLATFORM -MD -MT esp-idf/main/CMakeFiles/__idf_main.dir/app_main.c.obj -MF esp-idf/main/CMakeFiles/__idf_main.dir/app_main.c.obj.d -o esp-idf/main/CMakeFiles/__idf_main.dir/app_main.c.obj   -c ../main/app_main.c
../main/app_main.c:39:10: fatal error: app_wifi.h: No such file or directory
 #include <app_wifi.h>
          ^~~~~~~~~~~~
compilation terminated.
[1248/1311] Linking C static library esp-idf/wifi_provisioning/libwifi_provisioning.a
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

Thank you.