espressif/esp-homekit-sdk #47: Check whether is in pairing process

twlawrenceko, 2021-05-17:
Hello,

As pairing on ESP8266 is using quite much processing power, and as my program needs to monitor the gpio state continuously (i.e. using while loop), the monitor task will kind of disturbing the HomeKit pairing process which leads to failure (although setting lower priority than HomeKit task).

Is there any variable that can be tracked whether it is pairing with Home app or not?

Below is the pairing process serial monitor:
```
######## Starting Pair Setup ########
Pair Setup M1 Received
Pair Setup M2 Successful
Cleaning Pair Setup Context
######## Starting Pair Setup ########
Pair Setup M1 Received
Pair Setup M2 Successful
Cleaning Pair Setup Context
######## Starting Pair Setup ########
Pair Setup M1 Received
Pair Setup M2 Successful
Cleaning Pair Setup Context
######## Starting Pair Setup ########
Pair Setup M1 Received
Pair Setup M2 Successful
Cleaning Pair Setup Context
######## Starting Pair Setup ########
Pair Setup M1 Received
Pair Setup M2 Successful
Pair Setup M3 Received
Using pair-setup without MFi.
Pair Setup M4 Successful
Pair Setup M5 Received
Pair Setup Successful for FB660B62-3EF5-4F48-9848-4321FDF7B8D5
Updated state number to 2
Re-announcing _hap._tcp mDNS service
Cleaning Pair Setup Context
Cleaning Pair Setup Context
######## Starting Pair Verify ########
Pair Verify M1 Received
Pair Verify M2 Successful
Pair Verify M3 Received
HomeKit Session active
Pair Verify Successful for FB660B62-3EF5-4F48-9848-4321FDF7B8D5
Events Enabled for aid=1 iid=13
Events Enabled for aid=1 iid=16
Events Enabled for aid=1 iid=13
Events Enabled for aid=1 iid=16
Add Pairing received
Added Controller 96A1340C-6F2E-425E-9287-51090BB7A174
I (75784) HAP lightswitch: Accessory identified
```

It seems like `######## Starting Pair Setup ########` indicates it is now in pairing process and `I (75784) HAP lightswitch: Accessory identified` shows the pairing process is mostly done.

Thank you,
Lawrence

shahpiyushv, 2021-05-17:
@twlawrenceko , you may use the `HAP_EVENT_PAIRING_STARTED`,  `HAP_EVENT_CTRL_PAIRED` and `HAP_EVENT_PAIRING_ABORTED ` events to track the pair setup process. You can find sample usage in [Fan example](https://github.com/espressif/esp-homekit-sdk/blob/c62f64dea6669547182e932dfded0a3a912a1951/examples/fan/main/app_main.c#L105). However, note that, for ESP8266, I have noticed that pair setup indeed fails sometimes, when done via QR code but goes fine if the pin is provided manually (the delay during typing gives ESP8266 some additional time to process pair setup M1 message). Does your observation match this?

twlawrenceko, 2021-05-17:
@shahpiyushv , thank you for the tip. The pairing process with ESP8266 seems to be unstable, not for outlet identity but for lightbulb and light switch (based on my experience). I have not notice the big difference between using qrcode and pin, however, I will put this in mind.

I do have an additional question that I believe is quite related, as my monitor task is a while loop and if the wifi gets disconnected, I would also like to stop the monitor task, what is the variable for wifi not connected?

Thank you.