espressif/esp-homekit-sdk #51: lightbulb is not Paired

exrector, 2021-06-03:
rst:0x1 (POWERON_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6740
load:0x40078000,len:14240
ho 0 tail 12 room 4
load:0x40080400,len:3500
0x40080400: _init at ??:?

entry 0x4008064c
I (28) boot: ESP-IDF v4.4-dev-1404-gc13afea63 2nd stage bootloader
I (28) boot: compile time 01:13:37
I (28) boot: chip revision: 1
I (32) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (42) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 4MB
I (53) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 sec_cert         unknown          3f 00 0000d000 00003000
I (77) boot:  1 nvs              WiFi data        01 02 00010000 00006000
I (84) boot:  2 otadata          OTA data         01 00 00016000 00002000
I (92) boot:  3 phy_init         RF data          01 01 00018000 00001000
I (99) boot:  4 ota_0            OTA app          00 10 00020000 00190000
I (107) boot:  5 ota_1            OTA app          00 11 001b0000 00190000
I (114) boot:  6 factory_nvs      WiFi data        01 02 00340000 00006000
I (122) boot:  7 nvs_keys         NVS keys         01 04 00346000 00001000
I (130) boot: End of partition table
I (134) boot: No factory image, trying OTA 0
I (139) boot_comm: chip revision: 1, min. application chip revision: 0
I (146) esp_image: segment 0: paddr=00020020 vaddr=3f400020 size=428c0h (272576) map
I (253) esp_image: segment 1: paddr=000628e8 vaddr=3ffbdb60 size=0467ch ( 18044) load
I (260) esp_image: segment 2: paddr=00066f6c vaddr=40080000 size=090ach ( 37036) load
I (276) esp_image: segment 3: paddr=00070020 vaddr=400d0020 size=115e80h (1138304) map
I (688) esp_image: segment 4: paddr=00185ea8 vaddr=400890ac size=12d44h ( 77124) load
I (720) esp_image: segment 5: paddr=00198bf4 vaddr=50000000 size=00010h (    16) load
I (734) boot: Loaded app from partition at offset 0x20000
I (769) boot: Set actual ota_seq=1 in otadata[0]
I (769) boot: Disabling RNG early entropy source...
I (780) cpu_start: Pro cpu up.
I (780) cpu_start: Single core mode
I (791) cpu_start: Pro cpu start user code
I (791) cpu_start: cpu freq: 160000000
I (791) cpu_start: Application information:
I (795) cpu_start: Project name:     lightbulb
I (801) cpu_start: App version:      c62f64d
I (805) cpu_start: Compile time:     Jun  3 2021 01:13:20
I (812) cpu_start: ELF file SHA256:  35da324a148fc259...
I (818) cpu_start: ESP-IDF:          v4.4-dev-1404-gc13afea63
I (824) heap_init: Initializing. RAM available for dynamic allocation:
I (831) heap_init: At 3FF80000 len 00002000 (8 KiB): RTCRAM
I (837) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (844) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (850) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (856) heap_init: At 3FFCA8B8 len 00015748 (85 KiB): DRAM
I (862) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM
I (868) heap_init: At 40078000 len 00008000 (32 KiB): IRAM
I (875) heap_init: At 4009BDF0 len 00004210 (16 KiB): IRAM
I (882) spi_flash: detected chip: gd
I (885) spi_flash: flash io: dio
I (890) cpu_start: Starting scheduler on PRO CPU.
Keystore initialised
Accessory is not Paired with any controller
Database initialised. Accessory Device ID: ##:4F:##:92:F6:##
HAP Initialization succeeded. Version : 4.0-c62f64d
E (937) ledc: freq_hz=5000 duty_resolution=0

shahpiyushv, 2021-06-03:
@exrector , can you elaborate on the issue? Did you have hard-coded Wi-Fi credentials? If provisioning was enabled, did you provision the accessory? At what stage did you get the issue? Can you share the exact error seen or a screenshot of the Home app where you possibly got this error?

exrector, 2021-06-03:
@shahpiyushv 
this is the code from the last step - flashing.  all default values.  There are two qr codes.  The pairing process has started, but after a while homeapp gives an error.  I created a pair using esp ble prov.  Only.