espressif/esp-homekit-sdk #53: Setup ID absent

arthur-spa, 2021-06-04:
When following the "HomeKit Setup Configuration", using ./factory_nvs_gen.py and esptool.py, I have an error, when the esp8266 finished reset : 
[image]
Am I doing something wrong ?

shahpiyushv, 2021-08-04:
@arthur-spa ,sorry for the late reply? Were you able to go beyond this? Can you tell us the exact steps you followed? Can you share the binary generated using the `factory_nvs_gen` tool?

IsukaChiyogami, 2022-06-30:
I have the same problem ， could you share how you solve ？

IsukaChiyogami, 2022-06-30:
I got it in the source code (components\homekit\esp_hap_core\src\esp_hap_setup_payload.c)
```
    if (strlen(setup_code) != 10 || strlen(setup_id) != 4) {
        ESP_LOGE(TAG, "Setup code or Setup ID not correct. Eg. 111-22-333, ES32");
```
the lenght of setup_id should be 4

shahpiyushv, 2022-09-27:
Closing this issue as there was no update on this. This seems to be an issue of not flashing the factory_nvs binary at correct location

closed 2022-09-27 by shahpiyushv