espressif/esp-homekit-sdk #54: ESP32 HomeKit Session terminated problem.

lexwm, 2021-06-16:
Hello. I have problem on ESP32. I use latest versions esp-homekit-sdk and esp-idf. Python 3.9.4. macOS Big Sur 11.4. Processor M1.
I took a simple example of a light bulb as it is (without corrections on my part). Compiled / flash ESP32 successful. Without any errors / warnings. The first few minutes of work, everything is fine. After about 1 minutes, the connection is broken and the chip does not respond. 
Config and other options set by default.

```
I (2071) esp_netif_handlers: sta ip: 192.168.1.12, mask: 255.255.255.0, gw: 192.168.1.1
I (2071) app_wifi: Connected with IP Address:192.168.1.12
I (3071) app_wifi: Connected with IPv6 Address:fe80:0000:0000:0000:xxxx:xxxx:xxxx:xxxx
######## Starting Pair Verify ########
Pair Verify M1 Received
Pair Verify M2 Successful
Pair Verify M3 Received
HomeKit Session active. 
Pair Verify Successful for 8FB5698D-3270-4F4E-800E-87EB4D55CBBF
Enabling Keep-Alive on Pair Verify Session
######## Starting Pair Verify ########
Pair Verify M1 Received
Pair Verify M2 Successful
######## Starting Pair Verify ########
Pair Verify M1 Received
Pair Verify M2 Successful
Pair Verify M3 Received
HomeKit Session active. 
Pair Verify Successful for 00D0907C-D100-49B8-83A3-9BED625AFFC6
Enabling Keep-Alive on Pair Verify Session
Events Enabled for aid=1 iid=13
Events Enabled for aid=1 iid=15
Events Enabled for aid=1 iid=16
Events Enabled for aid=1 iid=17
Events Enabled for aid=1 iid=13
Events Enabled for aid=1 iid=15
Events Enabled for aid=1 iid=16
Events Enabled for aid=1 iid=17
Pair Verify M3 Received
HomeKit Session active. 
Pair Verify Successful for 8FB5698D-3270-4F4E-800E-87EB4D55CBBF
Enabling Keep-Alive on Pair Verify Session
Events Enabled for aid=1 iid=13
Events Enabled for aid=1 iid=15
Events Enabled for aid=1 iid=16
Events Enabled for aid=1 iid=17
Events Enabled for aid=1 iid=13
Events Enabled for aid=1 iid=15
Events Enabled for aid=1 iid=16
Events Enabled for aid=1 iid=17
I (45511) HAP lightbulb: Received Write for Light Off
I (45511) lightbulb: lightbulb_set_on : false
Value Changed
Notification Sent
HomeKit Session terminated.
I (155991) HAP lightbulb: Received Write for Light Off
I (155991) lightbulb: lightbulb_set_on : false
Value Changed
Notification Sent
I (157431) HAP lightbulb: Received Write for Light On
I (157431) lightbulb: lightbulb_set_on : true
Value Changed
Notification Sent
I (165421) HAP lightbulb: Received Write for Light Brightness 85
I (165421) lightbulb: lightbulb_set_brightness : 85
Value Changed
I (165421) HAP lightbulb: Received Write for Light On
I (165431) lightbulb: lightbulb_set_on : true
Notification Sent
I (165721) HAP lightbulb: Received Write for Light Brightness 86
I (165721) lightbulb: lightbulb_set_brightness : 86
Value Changed
I (165731) HAP lightbulb: Received Write for Light On
I (165731) lightbulb: lightbulb_set_on : true
Notification Sent
I (165921) HAP lightbulb: Received Write for Light Brightness 85
I (165931) lightbulb: lightbulb_set_brightness : 85
Value Changed
I (165931) HAP lightbulb: Received Write for Light On
I (165931) lightbulb: lightbulb_set_on : true
Notification Sent
HomeKit Session terminated.
HomeKit Session terminated.
```

This is about one minute of work of the chip. After last  Session terminated chip don’t respond. Any logs. In Apple Home Application - device shows as not available. After about 10 minutes awaiting chip start new session.

```
######## Starting Pair Verify ########
Pair Verify M1 Received
Pair Verify M2 Successful
######## Starting Pair Verify ########
Pair Verify M1 Received
Pair Verify M2 Successful
Pair Verify M3 Received
HomeKit Session active. Socket fd: 0
Pair Verify Successful for 00D0907C-D100-49B8-83A3-9BED625AFFC6
Enabling Keep-Alive on Pair Verify Session
######## Starting Pair Verify ########
Pair Verify M1 Received
Pair Verify M2 Successful
Pair Verify M3 Received
HomeKit Session active. Socket fd: 0
Pair Verify Successful for 8FB5698D-3270-4F4E-800E-87EB4D55CBBF
Enabling Keep-Alive on Pair Verify Session
######## Starting Pair Verify ########
Pair Verify M1 Received
Pair Verify M2 Successful
Pair Verify M3 Received
HomeKit Session active. Socket fd: 0
Pair Verify Successful for 8FB5698D-3270-4F4E-800E-87EB4D55CBBF
Enabling Keep-Alive on Pair Verify Session
Events Enabled for aid=1 iid=13
Events Enabled for aid=1 iid=15
Events Enabled for aid=1 iid=16
Events Enabled for aid=1 iid=17
Pair Verify M3 Received
HomeKit Session active. Socket fd: 0
Pair Verify Successful for 00D0907C-D100-49B8-83A3-9BED625AFFC6
Enabling Keep-Alive on Pair Verify Session
Events Enabled for aid=1 iid=13
Events Enabled for aid=1 iid=15
Events Enabled for aid=1 iid=16
```

It works again for about one minute і hangs again. So the process is repeated endlessly.
Please tell me what could I have done wrong? What i needs to be checked?

shahpiyushv, 2021-08-04:
@lexwm , do you see all these 3 lines together?

```
Notification Sent
HomeKit Session terminated.
HomeKit Session terminated.
```

Or does `HomeKit Session terminated` occur some time later?

HomeKit Session Terminated typically occurs when the controller closes the connection. Else, you would see the "Decryption error/Connection Lost" message.

lexwm, 2021-08-06:
it's all clear. 
I do not understand why the controller hangs and does not respond long enough after ?

Staars, 2021-10-05:
I've seen the same error message and behavior.

In my case the problem was caused by my Ubiquiti Aircube AC. An upgrade of the firmware solved it and I had not a single disconnect after that.
So check your network environment.