espressif/esp-homekit-sdk #56: bluetooth support?

societyofrobots, 2021-06-29:
Looking to implement a bluetooth device for iOS, but there doesn't seem to be any bluetooth support for ESP Homekit.

Am I missing something?

shahpiyushv, 2021-06-29:
@societyofrobots , the esp-homekit-sdk does not support bluetooth, and we don't have any plans to add it in the near future. Only Wi-Fi and Ethernet (basically IP transport) are supported.

societyofrobots, 2021-06-29:
Thanks, that's what I assumed.

It boggles the mind why Espressif didn't include bluetooth support for Homekit.

shahpiyushv, 2021-06-29:
@societyofrobots the primary reason was that ESP32s were best suited for IP based HomeKit accessories or for Wi-Fi - BLE HomeKit bridges and that's where most of the customer interest was.