espressif/esp-homekit-sdk #59: Anyone tried multiple devices/switches in Homekit example (MEGH-2692)

adukale, 2021-08-04:
I want to integrate multiple devices/switches in Homekit example code. The homekit initialization is clear to me but I am getting confused while registering devices ans setting callbacks, so wondering if anyone has already tried this.

Thanks in advance.
Any help is appreciated.

shahpiyushv, 2021-08-04:
@adukale , you can check the [bridge example](https://github.com/espressif/esp-homekit-sdk/blob/master/examples/bridge/main/app_main.c) to understand how to add multiple accessories under the same node. However, if the multiple accessories (switches in your case) are co-located i.e. part of a single physical device, you just need to add multiple switch services into the same accessory. You can either have independent callbacks for each or add some private data to each switch service (some identifier or GPIO number) and have a common callback. [This line](https://github.com/espressif/esp-homekit-sdk/blob/c62f64dea6669547182e932dfded0a3a912a1951/examples/bridge/main/app_main.c#L204) shows how to add private data and that same is used [here in the callback](https://github.com/espressif/esp-homekit-sdk/blob/c62f64dea6669547182e932dfded0a3a912a1951/examples/bridge/main/app_main.c#L128).

adukale, 2021-08-04:
@shahpiyushv Thank you for quick response.
Your suggestions helped. I have modified esp_rainmaker/example/homekit_example to have 6 switches.
Here is the code sample: [](https://github.com/adukale/homekit_6switch)

I can see 6 switches in Home app in ios/HomeKit. But turning on/off anyone of them results in rebooting esp32 and it goes into bootloop sometimes. Also, the connected gpios/relayes does not change the state.

Thanks in advance.