espressif/esp-homekit-sdk #60: Error in Wi-Fi provisioning

sateliermartin, 2021-08-06:
When executing the examples and trying to provision the WIFI, the Espressif Provisioning Apps find the bluetooth device. 

```
entry 0x40080664
I (27) boot: ESP-IDF v4.4-dev-2487-g8131d6f46-dirty 2nd stage bootloader
I (27) boot: compile time 23:50:41
I (28) boot: chip revision: 1
I (32) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 4MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 sec_cert         unknown          3f 00 0000d000 00003000
I (76) boot:  1 nvs              WiFi data        01 02 00010000 00006000
I (84) boot:  2 otadata          OTA data         01 00 00016000 00002000
I (91) boot:  3 phy_init         RF data          01 01 00018000 00001000
I (99) boot:  4 ota_0            OTA app          00 10 00020000 00190000
I (106) boot:  5 ota_1            OTA app          00 11 001b0000 00190000
I (114) boot:  6 factory_nvs      WiFi data        01 02 00340000 00006000
I (121) boot:  7 nvs_keys         NVS keys         01 04 00346000 00001000
I (129) boot: End of partition table
I (133) boot_comm: chip revision: 1, min. application chip revision: 0
I (140) esp_image: segment 0: paddr=00020020 vaddr=3f400020 size=3ec58h (257112) map
I (242) esp_image: segment 1: paddr=0005ec80 vaddr=3ffbdb60 size=01398h (  5016) load
I (244) esp_image: segment 2: paddr=00060020 vaddr=400d0020 size=115310h (1135376) map
I (659) esp_image: segment 3: paddr=00175338 vaddr=3ffbeef8 size=02fe8h ( 12264) load
I (664) esp_image: segment 4: paddr=00178328 vaddr=40080000 size=1b8ach (112812) load
I (711) esp_image: segment 5: paddr=00193bdc vaddr=50000000 size=00010h (    16) load
I (725) boot: Loaded app from partition at offset 0x20000
I (725) boot: Disabling RNG early entropy source...
I (736) cpu_start: Pro cpu up.
I (736) cpu_start: Single core mode
I (747) cpu_start: Pro cpu start user code
I (747) cpu_start: cpu freq: 160000000
I (747) cpu_start: Application information:
I (751) cpu_start: Project name:     lightbulb
I (756) cpu_start: App version:      c62f64d
I (761) cpu_start: Compile time:     Aug  5 2021 23:50:14
I (767) cpu_start: ELF file SHA256:  745e64b142ad374f...
I (773) cpu_start: ESP-IDF:          v4.4-dev-2487-g8131d6f46-dirty
I (781) heap_init: Initializing. RAM available for dynamic allocation:
I (788) heap_init: At 3FF80000 len 00002000 (8 KiB): RTCRAM
I (794) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (800) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (806) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (812) heap_init: At 3FFCA6B8 len 00015948 (86 KiB): DRAM
I (818) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM
I (825) heap_init: At 40078000 len 00008000 (32 KiB): IRAM
I (831) heap_init: At 4009B8AC len 00004754 (17 KiB): IRAM
I (838) spi_flash: detected chip: generic
I (842) spi_flash: flash io: dio
I (847) cpu_start: Starting scheduler on PRO CPU.
Keystore initialised
Accessory is not Paired with any controller
Database initialised. Accessory Device ID: 0A:60:FA:B4:26:DF
HAP Initialization succeeded. Version : 4.0-c62f64d
E (914) ledc: freq_hz=5000 duty_resolution=0
I (914) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3
I (924) app_hap_setup_payload: -----QR Code for HomeKit-----
I (934) app_hap_setup_payload: Scan this QR code from the Home app on iOS

  █▀▀▀▀▀█   █▀  █▀▀▀▀▀█
  █ ███ █ █▀▄▀▀ █ ███ █
  █ ▀▀▀ █ ██▀ █ █ ▀▀▀ █
  ▀▀▀▀▀▀▀ █ █ ▀ ▀▀▀▀▀▀▀
  ▀ ▀███▀ ▄▀ ▄█▄██▀██ ▄
  █▄█▄▀▄▀▀▀█▀▄█  ▀  ▄█▄
    ▀▀  ▀▀▄▄▀ ▀▄▀▄ ▄▄█▀
  █▀▀▀▀▀█ ▄  ▄█▀▄█ █ ▀▄
  █ ███ █ █▀█  ▄▀▀█▀  ▀
  █ ▀▀▀ █ ▀█▄ █▄ ▄ █▀ ▄
  ▀▀▀▀▀▀▀ ▀▀ ▀▀▀  ▀▀

I (1014) app_hap_setup_payload: If QR code is not visible, copy paste the below URL in a browser.
https://espressif.github.io/esp-homekit-sdk/qrcode.html?data=X-HM://00527813XES32
MFi auth not supported. Falling back to HAP_MFI_AUTH_NONE
I (1054) wifi:wifi driver task: 3ffcee9c, prio:23, stack:6656, core=0
I (1054) system_api: Base MAC address is not set
I (1054) system_api: read default base MAC address from EFUSE
I (1064) wifi:wifi firmware version: e381636
I (1064) wifi:wifi certification version: v7.0
I (1064) wifi:config NVS flash: enabled
I (1074) wifi:config nano formating: disabled
I (1074) wifi:Init data frame dynamic rx buffer num: 32
I (1074) wifi:Init management frame dynamic rx buffer num: 32
I (1084) wifi:Init management short buffer num: 32
I (1084) wifi:Init dynamic tx buffer num: 32
I (1094) wifi:Init static rx buffer size: 1600
I (1094) wifi:Init static rx buffer num: 10
I (1104) wifi:Init dynamic rx buffer num: 32
I (1104) wifi_init: rx ba win: 6
I (1104) wifi_init: tcpip mbox: 32
I (1114) wifi_init: udp mbox: 10
I (1114) wifi_init: tcp mbox: 6
I (1124) wifi_init: tcp tx win: 5744
I (1124) wifi_init: tcp rx win: 5744
I (1134) wifi_init: tcp mss: 1440
I (1134) wifi_init: WiFi IRAM OP enabled
I (1134) wifi_init: WiFi RX IRAM OP enabled
Setup ID: ES32
HAP Main Loop Started
mDNS initialised
Registering HomeKit web handlers
Announcing _hap._tcp mDNS service
I (1164) wifi_prov_scheme_ble: BT memory released
I (1164) app_wifi: Starting provisioning
I (1184) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (1264) wifi:mode : sta (3c:61:05:13:54:d8)
I (1264) wifi:enable tsf
I (1274) BTDM_INIT: BT controller compile version [d83d996]
I (1574) wifi_prov_mgr: Provisioning started with service name : PROV_1354D8
I (1574) app_wifi: Provisioning started
I (1574) app_wifi: -----QR Code for ESP Provisioning-----
I (1584) app_wifi: Scan this QR code from the phone app for Provisioning.

  █▀▀▀▀▀█ ▀▀    ▀ ▀  ▀ ▀███ █▀▀▀▀▀█
  █ ███ █ █▄ █▀█ ██▄█▀▄  ▀  █ ███ █
  █ ▀▀▀ █ ▀█▀▀▄ ▀▀▀▀███ ▀ ▄ █ ▀▀▀ █
  ▀▀▀▀▀▀▀ ▀▄▀▄▀ █▄█▄█ ▀ ▀▄▀ ▀▀▀▀▀▀▀
  ▀▀██ ▄▀▄█▄   ▄▄▀█▀ █▀▄ ▄█▀▄ ▀▀█▄█
  ▀█ ▀▀█▀█   ▄▄█▀ ▄▀▀▄▀ ███▀ ▀▄▀▀▀▀
  ▄▀▀▄▀▀▀▄██▄█ ▀▄███ ▄█▄▄█▄▀ █▀▄ ▀
   ▀▄▄ ▄▀ █   █     ██ ▄▀▄ ▄  ▄ █ ▄
   ███ █▀▄▀▄▀▀▄█ ▄▀▄▄ ▀▀▄▀█▄█▄█ █▀█
   ▄▄  ▀▀▀▄▄▀▀█▀▄█▄▄▀▄▄   ▄  ▀▀▄ ▀
  ▄▄ ▀▄▄▀ ▄▀ ▀█▄▀▀█▀▄▄▀▀▀▀ █▄▄  █▀▄
   ▄██ ▄▀▀▄▄█▀▀▄  █   ▀ ██ ▄▄▄ █▀ █
  ▀ ▀ ▀▀▀ █ ▀▄█▀██▄▀▀▄▄   █▀▀▀█▀ ▄▀
  █▀▀▀▀▀█  ▄██ ▀█ ▀ █▀ █▀▄█ ▀ █ █
  █ ███ █ ▄▀▀  █▀▄▀   ▄▀▀▀▀█▀▀▀ ▀█▀
  █ ▀▀▀ █ █ ▀▄▄▀▄▀▄█▀▄▀▀  ▀ █▀▄█▀▀▄
  ▀▀▀▀▀▀▀ ▀ ▀▀▀ ▀  ▀▀  ▀ ▀ ▀▀ ▀ ▀

I (1794) app_wifi: If QR code is not visible, copy paste the below URL in a browser.
https://espressif.github.io/esp-jumpstart/qrcode.html?data={"ver":"v1","name":"PROV_1354D8","pop":"051354d8","transport":"ble"}
I (1814) app_wifi: Provisioning Started. Name : PROV_1354D8, POP : 051354d8
```

In my iPhone, not found

[image]

Use a lightbulb example in repo.

vikas-chandra-mnnit, 2021-08-06:
Hi @sateliermartin, can you make sure that the BLE device is in the discoverable state and is not connected with any other apps on your iPhone. Also please provide us with these additional information so we can try this out at our end:

1. App version.
2. iOS version.
3. Frequency of issue.

sateliermartin, 2021-08-06:
@vikas-chandra-mnnit thanks for the reply.

BLE Config:
```
#
# Bluetooth controller(ESP32 Dual Mode Bluetooth)
#
CONFIG_BTDM_CTRL_MODE_BLE_ONLY=y
# CONFIG_BTDM_CTRL_MODE_BR_EDR_ONLY is not set
# CONFIG_BTDM_CTRL_MODE_BTDM is not set
CONFIG_BTDM_CTRL_BLE_MAX_CONN=3
CONFIG_BTDM_CTRL_BR_EDR_SCO_DATA_PATH_EFF=0
CONFIG_BTDM_CTRL_PCM_ROLE_EFF=0
CONFIG_BTDM_CTRL_PCM_POLAR_EFF=0
CONFIG_BTDM_CTRL_BLE_MAX_CONN_EFF=3
CONFIG_BTDM_CTRL_BR_EDR_MAX_ACL_CONN_EFF=0
CONFIG_BTDM_CTRL_BR_EDR_MAX_SYNC_CONN_EFF=0
CONFIG_BTDM_CTRL_PINNED_TO_CORE=0
CONFIG_BTDM_CTRL_HCI_MODE_VHCI=y
# CONFIG_BTDM_CTRL_HCI_MODE_UART_H4 is not set

#
# MODEM SLEEP Options
#
CONFIG_BTDM_CTRL_MODEM_SLEEP=y
CONFIG_BTDM_CTRL_MODEM_SLEEP_MODE_ORIG=y
# CONFIG_BTDM_CTRL_MODEM_SLEEP_MODE_EVED is not set
CONFIG_BTDM_CTRL_LPCLK_SEL_MAIN_XTAL=y
# end of MODEM SLEEP Options

CONFIG_BTDM_BLE_DEFAULT_SCA_250PPM=y
CONFIG_BTDM_BLE_SLEEP_CLOCK_ACCURACY_INDEX_EFF=1
CONFIG_BTDM_BLE_SCAN_DUPL=y
CONFIG_BTDM_SCAN_DUPL_TYPE_DEVICE=y
# CONFIG_BTDM_SCAN_DUPL_TYPE_DATA is not set
# CONFIG_BTDM_SCAN_DUPL_TYPE_DATA_DEVICE is not set
CONFIG_BTDM_SCAN_DUPL_TYPE=0
CONFIG_BTDM_SCAN_DUPL_CACHE_SIZE=200
# CONFIG_BTDM_BLE_MESH_SCAN_DUPL_EN is not set
CONFIG_BTDM_CTRL_FULL_SCAN_SUPPORTED=y
CONFIG_BTDM_BLE_ADV_REPORT_FLOW_CTRL_SUPP=y
CONFIG_BTDM_BLE_ADV_REPORT_FLOW_CTRL_NUM=100
CONFIG_BTDM_BLE_ADV_REPORT_DISCARD_THRSHOLD=20
# end of Bluetooth controller(ESP32 Dual Mode Bluetooth)
```

1. App version v2.0.11 (5485c79)
2. IOS version 14.7.1 - iPhone 11 Pro

it is not connected with another application.

On my iPhone on Mac they don't see the device but my Windows PC yes.

lorenzofiamingo, 2021-08-10:
Same error here. I think this is because esp32 hasn't a device name. How can we set this?

vikas-chandra-mnnit, 2021-08-10:
@sateliermartin We tried replicating the behaviour using:

1. App version v2.0.11 (5485c79)
2. iOS version 14.7.1
3. ESP-IDF v4.4-dev

We did not found any issue and provisioning was successful. If you have made any changes in the ESP-IDF source that you think might be relevant to this issue, please let us know. 

We will continue to look into this. Meanwhile, I would suggest that you try some basic steps like restarting the iPhone or re-installing the provisioning apps to check if it works.

Suhas7, 2021-08-22:
I'm also having this issue, flashed on an adafruit esp32 feather. I flashed the vanilla source repo, no adjustments and clean idf install.

vikas-chandra-mnnit, 2021-09-08:
Hi @sateliermartin  @Suhas7, we have tracked down the issue and it may be related to the pairing of BLE devices that was introduced later in the ESP-IDF. Due to this connection with BLE device is successful for the first time but if we do `erase_flash` on the same device then pairing information is getting erased from the device. However, iOS still keeps this information and tries to pair it with that which ultimately leads to failure.

We are currently working on the fix. For now, you can go to `iPhone->Settings->Bluetooth` and forget the device from the list and try again. Let us know if doing this fixes your issue.

sateliermartin, 2021-09-08:
Hi @vikas-chandra-mnnit, I can't pair the device for the first time so I don't have it on the mentioned list.

sateliermartin, 2022-01-31:
@wangyz1997 thanks, but it didn't work for me.

Could you copy sdkconfg file? I put a copy of mine. 

```
#
# Bluedroid Options
#
CONFIG_BT_BTC_TASK_STACK_SIZE=3072
CONFIG_BT_BLUEDROID_PINNED_TO_CORE=0
CONFIG_BT_BTU_TASK_STACK_SIZE=4096
# CONFIG_BT_BLUEDROID_MEM_DEBUG is not set
# CONFIG_BT_CLASSIC_ENABLED is not set
CONFIG_BT_BLE_ENABLED=y
CONFIG_BT_GATTS_ENABLE=y
# CONFIG_BT_GATTS_PPCP_CHAR_GAP is not set
CONFIG_BT_BLE_BLUFI_ENABLE=y
CONFIG_BT_GATT_MAX_SR_PROFILES=8
# CONFIG_BT_GATTS_SEND_SERVICE_CHANGE_MANUAL is not set
CONFIG_BT_GATTS_SEND_SERVICE_CHANGE_AUTO=y
CONFIG_BT_GATTS_SEND_SERVICE_CHANGE_MODE=0
CONFIG_BT_GATTC_ENABLE=y
# CONFIG_BT_GATTC_CACHE_NVS_FLASH is not set
CONFIG_BT_GATTC_CONNECT_RETRY_COUNT=3
CONFIG_BT_BLE_SMP_ENABLE=y
# CONFIG_BT_SMP_SLAVE_CON_PARAMS_UPD_ENABLE is not set
# CONFIG_BT_STACK_NO_LOG is not set
```