espressif/esp-homekit-sdk #67: ninja.build file garbled code

akakazi, 2021-09-05:
An error occurs when I using vscode expressif IDF ext build example under win10:
[image]

and when I check ninja.build, there are some garbled code in this line:
[image]

the config button also doesn't work:
[image]

 But build the example in esp-now is successful.
Should I change CMakeLists.txt or build it in others env?

GlenTrudgett, 2021-12-23:
> An error occurs when I using vscode expressif IDF ext build example under win10, It seems that there is a lexing error in this line: [image]
> 
> and when I check ninja.build, there are some garbled code in this line: [image]
> 
> But build the example in esp-now is successful. Should I change CMakeLists.txt or build it in others env?

same result as well. bumping!

Yinke7, 2023-02-02:
i has a similer trouble, do you deal it ?