espressif/esp-homekit-sdk #69: esp32c3 failed to build samples

pwallner, 2021-09-27:
I tried to build for esp32c3, but it fails (also tried esp32 and esp32s2 and they worked!):
```
[1245/1314] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/lightbulb.c.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/lightbulb.c.obj
/home/mcpat/.espressif/tools/riscv32-esp-elf/esp-2021r1-8.4.0/riscv32-esp-elf/bin/riscv32-esp-elf-gcc -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -Iconfig -I../main -I/home/mcpat/esp/esp-idf/components/newlib/platform_include -I/home/mcpat/esp/esp-idf/components/freertos/include -I/home/mcpat/esp/esp-idf/components/freertos/include/esp_additions/freertos -I/home/mcpat/esp/esp-idf/components/freertos/port/riscv/include -I/home/mcpat/esp/esp-idf/components/freertos/include/esp_additions -I/home/mcpat/esp/esp-idf/components/esp_hw_support/include -I/home/mcpat/esp/esp-idf/components/esp_hw_support/include/soc -I/home/mcpat/esp/esp-idf/components/esp_hw_support/include/soc/esp32c3 -I/home/mcpat/esp/esp-idf/components/esp_hw_support/port/esp32c3/. -I/home/mcpat/esp/esp-idf/components/esp_hw_support/port/esp32c3/private_include -I/home/mcpat/esp/esp-idf/components/heap/include -I/home/mcpat/esp/esp-idf/components/log/include -I/home/mcpat/esp/esp-idf/components/lwip/include/apps -I/home/mcpat/esp/esp-idf/components/lwip/include/apps/sntp -I/home/mcpat/esp/esp-idf/components/lwip/lwip/src/include -I/home/mcpat/esp/esp-idf/components/lwip/port/esp32/include -I/home/mcpat/esp/esp-idf/components/lwip/port/esp32/include/arch -I/home/mcpat/esp/esp-idf/components/soc/include -I/home/mcpat/esp/esp-idf/components/soc/esp32c3/. -I/home/mcpat/esp/esp-idf/components/soc/esp32c3/include -I/home/mcpat/esp/esp-idf/components/hal/esp32c3/include -I/home/mcpat/esp/esp-idf/components/hal/include -I/home/mcpat/esp/esp-idf/components/hal/platform_port/include -I/home/mcpat/esp/esp-idf/components/esp_rom/include -I/home/mcpat/esp/esp-idf/components/esp_rom/include/esp32c3 -I/home/mcpat/esp/esp-idf/components/esp_rom/esp32c3 -I/home/mcpat/esp/esp-idf/components/esp_common/include -I/home/mcpat/esp/esp-idf/components/esp_system/include -I/home/mcpat/esp/esp-idf/components/esp_system/port/soc -I/home/mcpat/esp/esp-idf/components/esp_system/port/include/riscv -I/home/mcpat/esp/esp-idf/components/esp_system/port/public_compat -I/home/mcpat/esp/esp-idf/components/riscv/include -I/home/mcpat/esp/esp-idf/components/driver/include -I/home/mcpat/esp/esp-idf/components/driver/esp32c3/include -I/home/mcpat/esp/esp-idf/components/esp_pm/include -I/home/mcpat/esp/esp-idf/components/esp_ringbuf/include -I/home/mcpat/esp/esp-idf/components/efuse/include -I/home/mcpat/esp/esp-idf/components/efuse/esp32c3/include -I/home/mcpat/esp/esp-idf/components/vfs/include -I/home/mcpat/esp/esp-idf/components/esp_wifi/include -I/home/mcpat/esp/esp-idf/components/esp_event/include -I/home/mcpat/esp/esp-idf/components/esp_netif/include -I/home/mcpat/esp/esp-idf/components/esp_eth/include -I/home/mcpat/esp/esp-idf/components/tcpip_adapter/include -I/home/mcpat/esp/esp-idf/components/esp_phy/include -I/home/mcpat/esp/esp-idf/components/esp_phy/esp32c3/include -I/home/mcpat/esp/esp-idf/components/esp_ipc/include -I/home/mcpat/esp/esp-idf/components/app_trace/include -I/home/mcpat/esp/esp-idf/components/esp_timer/include -I/home/mcpat/esp/esp-idf/components/mbedtls/port/include -I/home/mcpat/esp/esp-idf/components/mbedtls/mbedtls/include -I/home/mcpat/esp/esp-idf/components/mbedtls/esp_crt_bundle/include -I/home/mcpat/esp/esp-idf/components/app_update/include -I/home/mcpat/esp/esp-idf/components/spi_flash/include -I/home/mcpat/esp/esp-idf/components/bootloader_support/include -I/home/mcpat/esp/esp-idf/components/nvs_flash/include -I/home/mcpat/esp/esp-idf/components/pthread/include -I/home/mcpat/esp/esp-idf/components/esp_gdbstub/include -I/home/mcpat/esp/esp-idf/components/esp_gdbstub/riscv -I/home/mcpat/esp/esp-idf/components/esp_gdbstub/esp32c3 -I/home/mcpat/esp/esp-idf/components/espcoredump/include -I/home/mcpat/esp/esp-idf/components/espcoredump/include/port/riscv -I/home/mcpat/esp/esp-idf/components/wpa_supplicant/include -I/home/mcpat/esp/esp-idf/components/wpa_supplicant/port/include -I/home/mcpat/esp/esp-idf/components/wpa_supplicant/esp_supplicant/include -I/home/mcpat/esp/esp-idf/components/ieee802154/include -I/home/mcpat/esp/esp-idf/components/asio/asio/asio/include -I/home/mcpat/esp/esp-idf/components/asio/port/include -I/home/mcpat/esp/esp-idf/components/bt/common/osi/include -I/home/mcpat/esp/esp-idf/components/bt/include/esp32c3/include -I/home/mcpat/esp/esp-idf/components/bt/common/api/include/api -I/home/mcpat/esp/esp-idf/components/bt/common/btc/profile/esp/blufi/include -I/home/mcpat/esp/esp-idf/components/bt/common/btc/profile/esp/include -I/home/mcpat/esp/esp-idf/components/bt/host/bluedroid/api/include/api -I/home/mcpat/esp/esp-idf/components/cbor/port/include -I/home/mcpat/esp/esp-idf/components/unity/include -I/home/mcpat/esp/esp-idf/components/unity/unity/src -I/home/mcpat/esp/esp-idf/components/cmock/CMock/src -I/home/mcpat/esp/esp-idf/components/coap/port/include -I/home/mcpat/esp/esp-idf/components/coap/libcoap/include -I/home/mcpat/esp/esp-idf/components/console -I/home/mcpat/esp/esp-idf/components/nghttp/port/include -I/home/mcpat/esp/esp-idf/components/nghttp/nghttp2/lib/includes -I/home/mcpat/esp/esp-idf/components/esp-tls -I/home/mcpat/esp/esp-idf/components/esp-tls/esp-tls-crypto -I/home/mcpat/esp/esp-idf/components/esp_adc_cal/include -I/home/mcpat/esp/esp-idf/components/esp_hid/include -I/home/mcpat/esp/esp-idf/components/tcp_transport/include -I/home/mcpat/esp/esp-idf/components/esp_http_client/include -I/home/mcpat/esp/esp-idf/components/esp_http_server/include -I/home/mcpat/esp/esp-idf/components/esp_https_ota/include -I/home/mcpat/esp/esp-idf/components/esp_lcd/include -I/home/mcpat/esp/esp-idf/components/esp_lcd/interface -I/home/mcpat/esp/esp-idf/components/protobuf-c/protobuf-c -I/home/mcpat/esp/esp-idf/components/protocomm/include/common -I/home/mcpat/esp/esp-idf/components/protocomm/include/security -I/home/mcpat/esp/esp-idf/components/protocomm/include/transports -I/home/mcpat/esp/esp-idf/components/mdns/include -I/home/mcpat/esp/esp-idf/components/esp_local_ctrl/include -I/home/mcpat/esp/esp-idf/components/sdmmc/include -I/home/mcpat/esp/esp-idf/components/esp_serial_slave_link/include -I/home/mcpat/esp/esp-idf/components/esp_websocket_client/include -I/home/mcpat/esp/esp-idf/components/expat/expat/expat/lib -I/home/mcpat/esp/esp-idf/components/expat/port/include -I/home/mcpat/esp/esp-idf/components/wear_levelling/include -I/home/mcpat/esp/esp-idf/components/fatfs/diskio -I/home/mcpat/esp/esp-idf/components/fatfs/vfs -I/home/mcpat/esp/esp-idf/components/fatfs/src -I/home/mcpat/esp/esp-idf/components/freemodbus/common/include -I/home/mcpat/esp/esp-idf/components/idf_test/include -I/home/mcpat/esp/esp-idf/components/idf_test/include/esp32c3 -I/home/mcpat/esp/esp-idf/components/jsmn/include -I/home/mcpat/esp/esp-idf/components/json/cJSON -I/home/mcpat/esp/esp-idf/components/libsodium/libsodium/src/libsodium/include -I/home/mcpat/esp/esp-idf/components/libsodium/port_include -I/home/mcpat/esp/esp-idf/components/mqtt/esp-mqtt/include -I/home/mcpat/esp/esp-idf/components/openssl/include -I/home/mcpat/esp/esp-idf/components/spiffs/include -I/home/mcpat/esp/esp-idf/components/wifi_provisioning/include -I/home/mcpat/esp/esp-homekit-sdk/components/button/button/include -I/home/mcpat/esp/esp-homekit-sdk/components/homekit/hkdf-sha/include -I/home/mcpat/esp/esp-homekit-sdk/components/homekit/mu_srp -I/home/mcpat/esp/esp-homekit-sdk/components/homekit/json_generator/upstream -I/home/mcpat/esp/esp-homekit-sdk/components/homekit/json_parser/upstream/include -I/home/mcpat/esp/esp-homekit-sdk/components/homekit/json_parser/upstream -I/home/mcpat/esp/esp-homekit-sdk/components/homekit/esp_hap_platform/include -I/home/mcpat/esp/esp-homekit-sdk/components/homekit/esp_hap_core/include -I/home/mcpat/esp/esp-homekit-sdk/components/homekit/esp_hap_apple_profiles/include -I/home/mcpat/esp/esp-homekit-sdk/components/homekit/esp_hap_extras/include -I/home/mcpat/esp/esp-homekit-sdk/examples/common/qrcode/include -I/home/mcpat/esp/esp-homekit-sdk/examples/common/app_hap_setup_payload -I/home/mcpat/esp/esp-homekit-sdk/examples/common/app_wifi -march=rv32imc -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated- [... cut ...]
../main/lightbulb.c: In function 'lightbulb_deinit':
../main/lightbulb.c:213:15: error: 'LEDC_HIGH_SPEED_MODE' undeclared (first use in this function); did you mean 'LEDC_LOW_SPEED_MODE'?
     ledc_stop(LEDC_HIGH_SPEED_MODE, LEDC_CHANNEL_0, 0);
               ^~~~~~~~~~~~~~~~~~~~
               LEDC_LOW_SPEED_MODE
[1250/1314] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/app_main.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

I don't know whats going wrong...

Using Debian 11.

Thx for help,
Patrick

shahpiyushv, 2021-09-28:
@pwallner , the lightbulb example indeed does not compile for esp32c3 since the led driver does not seem compatible with esp32c3. However, the Fan example is compiling well. Can you double check that all your esp-idf git submodules are in sync by executing `git submodule update --init --recursive` in esp-idf? Also perform `./install.sh` and `. ./export.sh` once again in your esp-idf folder just to be sure that you have latest toolchain and correct environment set.

pwallner, 2021-09-28:
@shahpiyushv , I did now like written above

then
```
 idf.py set-target esp32c3
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Executing action: set-target
Set Target to: esp32c3, new sdkconfig created. Existing sdkconfig renamed to sdkconfig.old.
Running cmake in directory /home/mcpat/esp/esp-homekit-sdk/examples/fan2/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32c3 -DCCACHE_ENABLE=0 /home/mcpat/esp/esp-homekit-sdk/examples/fan2"...
-- Found Git: /usr/bin/git (found version "2.30.2")
-- Component directory /home/mcpat/esp/esp-homekit-sdk/components/homekit does not contain a CMakeLists.txt file. No component will be added
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/mcpat/.espressif/tools/riscv32-esp-elf/esp-2021r1-8.4.0/riscv32-esp-elf/bin/riscv32-esp-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/mcpat/.espressif/tools/riscv32-esp-elf/esp-2021r1-8.4.0/riscv32-esp-elf/bin/riscv32-esp-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/mcpat/.espressif/tools/riscv32-esp-elf/esp-2021r1-8.4.0/riscv32-esp-elf/bin/riscv32-esp-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32c3
-- Project sdkconfig file /home/mcpat/esp/esp-homekit-sdk/examples/fan2/sdkconfig
Loading defaults file /home/mcpat/esp/esp-homekit-sdk/examples/fan2/sdkconfig.defaults...
-- Found PythonInterp: /home/mcpat/.espressif/python_env/idf4.4_py3.9_env/bin/python (found version "3.9.2")
-- Found Perl: /usr/bin/perl (found version "5.32.1")
-- App "fan" version: c62f64d
-- Adding linker script /home/mcpat/esp/esp-homekit-sdk/examples/fan2/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /home/mcpat/esp/esp-idf/components/esp_system/ld/esp32c3/sections.ld.in
-- Adding linker script /home/mcpat/esp/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.ld
-- Adding linker script /home/mcpat/esp/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.api.ld
-- Adding linker script /home/mcpat/esp/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.libgcc.ld
-- Adding linker script /home/mcpat/esp/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.newlib.ld
-- Adding linker script /home/mcpat/esp/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.version.ld
-- Adding linker script /home/mcpat/esp/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.eco3.ld
-- Adding linker script /home/mcpat/esp/esp-idf/components/soc/esp32c3/ld/esp32c3.peripherals.ld
4.0-c62f64d
ESP HomeKit SDK Project commit: c62f64d
-- Components: app_hap_setup_payload app_trace app_update app_wifi asio bootloader bootloader_support bt button cbor cmock coap console cxx driver efuse esp-tls esp32c3 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hap_apple_profiles esp_hap_core esp_hap_extras esp_hap_platform esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_ipc esp_lcd esp_local_ctrl esp_netif esp_phy esp_pm esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos hal heap hkdf-sha idf_test ieee802154 jsmn json json_generator json_parser libsodium log lwip main mbedtls mdns mqtt mu_srp newlib nghttp nvs_flash openssl openthread partition_table protobuf-c protocomm pthread qrcode riscv sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter tinyusb unity usb vfs wear_levelling wifi_provisioning wpa_supplicant
-- Component paths: /home/mcpat/esp/esp-homekit-sdk/examples/common/app_hap_setup_payload /home/mcpat/esp/esp-idf/components/app_trace /home/mcpat/esp/esp-idf/components/app_update /home/mcpat/esp/esp-homekit-sdk/examples/common/app_wifi /home/mcpat/esp/esp-idf/components/asio /home/mcpat/esp/esp-idf/components/bootloader /home/mcpat/esp/esp-idf/components/bootloader_support /home/mcpat/esp/esp-idf/components/bt /home/mcpat/esp/esp-homekit-sdk/components/button /home/mcpat/esp/esp-idf/components/cbor /home/mcpat/esp/esp-idf/components/cmock /home/mcpat/esp/esp-idf/components/coap /home/mcpat/esp/esp-idf/components/console /home/mcpat/esp/esp-idf/components/cxx /home/mcpat/esp/esp-idf/components/driver /home/mcpat/esp/esp-idf/components/efuse /home/mcpat/esp/esp-idf/components/esp-tls /home/mcpat/esp/esp-idf/components/esp32c3 /home/mcpat/esp/esp-idf/components/esp_adc_cal /home/mcpat/esp/esp-idf/components/esp_common /home/mcpat/esp/esp-idf/components/esp_eth /home/mcpat/esp/esp-idf/components/esp_event /home/mcpat/esp/esp-idf/components/esp_gdbstub /home/mcpat/esp/esp-homekit-sdk/components/homekit/esp_hap_apple_profiles /home/mcpat/esp/esp-homekit-sdk/components/homekit/esp_hap_core /home/mcpat/esp/esp-homekit-sdk/components/homekit/esp_hap_extras /home/mcpat/esp/esp-homekit-sdk/components/homekit/esp_hap_platform /home/mcpat/esp/esp-idf/components/esp_hid /home/mcpat/esp/esp-idf/components/esp_http_client /home/mcpat/esp/esp-idf/components/esp_http_server /home/mcpat/esp/esp-idf/components/esp_https_ota /home/mcpat/esp/esp-idf/components/esp_https_server /home/mcpat/esp/esp-idf/compone [... cut ...]
/home/mcpat/.espressif/tools/riscv32-esp-elf/esp-2021r1-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /home/mcpat/esp/esp-idf/components/protocomm/src/simple_ble/simple_ble.c:111: undefined reference to `esp_ble_gap_config_scan_rsp_data_raw'
/home/mcpat/.espressif/tools/riscv32-esp-elf/esp-2021r1-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /home/mcpat/esp/esp-idf/components/protocomm/src/simple_ble/simple_ble.c:155: undefined reference to `esp_ble_gap_start_advertising'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

pwallner, 2021-10-12:
I have now more details, maybe this helps:
```
[1252/1313] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/app_main.c.obj
../main/app_main.c: In function 'fan_thread_entry':
../main/app_main.c:313:61: warning: passing argument 3 of 'esp_event_handler_register' from incompatible pointer type [-Wincompatible-pointer-types]
     esp_event_handler_register(HAP_EVENT, ESP_EVENT_ANY_ID, &fan_hap_event_handler, NULL);
                                                             ^~~~~~~~~~~~~~~~~~~~~~
In file included from ../main/app_main.c:32:
/home/mcpat/esp/esp-idf/components/esp_event/include/esp_event.h:153:58: note: expected 'esp_event_handler_t' {aka 'void (*)(void *, const char *, long int,  void *)'} but argument is of type 'void (*)(void *, const char *, int,  void *)'
                                      esp_event_handler_t event_handler,
                                      ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~
[1306/1313] Building C object esp-idf/app_wifi/CMakeFiles/__idf_app_wifi.dir/app_wifi.c.obj
In file included from /home/mcpat/esp/esp-idf/components/esp_timer/include/esp_timer.h:45,
                 from /home/mcpat/esp/esp-idf/components/freertos/port/riscv/include/freertos/portmacro.h:41,
                 from /home/mcpat/esp/esp-idf/components/freertos/include/freertos/portable.h:51,
                 from /home/mcpat/esp/esp-idf/components/freertos/include/freertos/FreeRTOS.h:63,
                 from /home/mcpat/esp/esp-homekit-sdk/examples/common/app_wifi/app_wifi.c:26:
/home/mcpat/esp/esp-homekit-sdk/examples/common/app_wifi/app_wifi.c: In function 'app_wifi_init':
/home/mcpat/esp/esp-homekit-sdk/examples/common/app_wifi/app_wifi.c:262:78: warning: passing argument 3 of 'esp_event_handler_register' from incompatible pointer type [-Wincompatible-pointer-types]
     ESP_ERROR_CHECK(esp_event_handler_register(WIFI_EVENT, ESP_EVENT_ANY_ID, &event_handler, wifi_netif));
                                                                              ^~~~~~~~~~~~~~
/home/mcpat/esp/esp-idf/components/esp_common/include/esp_err.h:122:30: note: in definition of macro 'ESP_ERROR_CHECK'
         esp_err_t err_rc_ = (x);                                        \
                              ^
In file included from /home/mcpat/esp/esp-idf/components/esp_wifi/include/esp_wifi.h:64,
                 from /home/mcpat/esp/esp-homekit-sdk/examples/common/app_wifi/app_wifi.c:29:
/home/mcpat/esp/esp-idf/components/esp_event/include/esp_event.h:153:58: note: expected 'esp_event_handler_t' {aka 'void (*)(void *, const char *, long int,  void *)'} but argument is of type 'void (*)(void *, const char *, int,  void *)'
                                      esp_event_handler_t event_handler,
                                      ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~
In file included from /home/mcpat/esp/esp-idf/components/esp_timer/include/esp_timer.h:45,
                 from /home/mcpat/esp/esp-idf/components/freertos/port/riscv/include/freertos/portmacro.h:41,
                 from /home/mcpat/esp/esp-idf/components/freertos/include/freertos/portable.h:51,
                 from /home/mcpat/esp/esp-idf/components/freertos/include/freertos/FreeRTOS.h:63,
                 from /home/mcpat/esp/esp-homekit-sdk/examples/common/app_wifi/app_wifi.c:26:
/home/mcpat/esp/esp-homekit-sdk/examples/common/app_wifi/app_wifi.c:266:79: warning: passing argument 3 of 'esp_event_handler_register' from incompatible pointer type [-Wincompatible-pointer-types]
     ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, IP_EVENT_STA_GOT_IP, &event_handler, NULL));
                                                                               ^~~~~~~~~~~~~~
/home/mcpat/esp/esp-idf/components/esp_common/include/esp_err.h:122:30: note: in definition of macro 'ESP_ERROR_CHECK'
         esp_err_t err_rc_ = (x);                                        \
                              ^
In file included from /home/mcpat/esp/esp-idf/components/esp_wifi/include/esp_wifi.h:64,
                 from /home/mcpat/esp/esp-homekit-sdk/examples/common/app_wifi/app_wifi.c:29:
/home/mcpat/esp/esp-idf/components/esp_event/include/esp_event.h:153:58: note: expected 'esp_event_handler_t' {aka 'void (*)(void *, const char *, long int,  void *)'} but argument is of type 'void (*)(void *, const char *, int,  void *)'
                                      esp_event_handler_t event_handler,
                                      ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~
In file included from /home/mcpat/esp/esp-idf/components/esp_timer/include/esp_timer.h:45,
                 from /home/mcpat/esp/esp-idf/components/freertos/port/riscv/include/freertos/portmacro.h:41,
                 from /home/mcpat/esp/esp-idf/components/freertos/include/freertos/portable.h:51,
                 from /home/mcpat/esp/esp-idf/components/freertos/include/freertos/FreeRTOS.h:63,
                 from /home/mcpat/esp/esp-homekit-sdk/examples/common/app_wifi/app_wifi.c:26:
/home/mcpat/esp/esp-homekit-sdk/examples/common/app_wifi/app_wifi.c:267:76: warning: passing argument 3 of 'esp_event_handler_register' from incompatible pointer type [-Wincompatible-pointer-types]
     ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, IP_EVENT_GOT_IP6, &event_handler, NULL));
                                                                            ^~~~~~~~~~~~~~
/home/mcpat/esp/esp-idf/components/esp_common/include/esp_err.h:122:30: note: in definition of macro 'ESP_ERROR_CHECK'
[... cut ...]
/home/mcpat/.espressif/tools/riscv32-esp-elf/esp-2021r1-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /home/mcpat/esp/esp-idf/components/protocomm/src/simple_ble/simple_ble.c:111: undefined reference to `esp_ble_gap_config_scan_rsp_data_raw'
/home/mcpat/.espressif/tools/riscv32-esp-elf/esp-2021r1-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /home/mcpat/esp/esp-idf/components/protocomm/src/simple_ble/simple_ble.c:155: undefined reference to `esp_ble_gap_start_advertising'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

pwallner, 2021-10-15:
Downloaded and installed everything for Windows (VS Code), this is working for fan example, but I have to change from BlueDroid - Dual-Mode to NimBLE - BLE only, otherwise I get errors. What's the difference?

Also try now to modify the lightbulb to include esp-idf example "led_strip". This example works with the embedded RGB LED on my ESP32-C3-DevKITM-1. Had also to modify this example a little bit (GPIO8 for LED and 1 Led in menuconfig).

leeebo, 2023-02-13:
This a type error, and has been fixed from [app_wifi.c#L163](https://github.com/espressif/esp-homekit-sdk/blob/master/examples/common/app_wifi/app_wifi.c#L163)