espressif/esp-homekit-sdk #72: Manual Lightbulb control with a switch

matheusesperanca, 2021-10-01:
Hi there,

I used Lightbulb example on ESP32 and everything is working.

I can't find some solution to get the handlers to change the light state manually (with a switch), how I can do this?

Thanks.

shahpiyushv, 2021-10-01:
@matheusesperanca , you can use the push button logic from [here](https://github.com/espressif/esp-homekit-sdk/blob/c62f64dea6669547182e932dfded0a3a912a1951/examples/lightbulb/main/app_main.c#L84) and register your callback.

Sample:

```c
button_handle_t btn_handle = iot_button_create(BUTTON_GPIO, BUTTON_ACTIVE_LEVEL);
if (btn_handle) {
    /* Register a callback for a button tap (short press) event */
    iot_button_set_evt_cb(btn_handle, BUTTON_CB_TAP, push_btn_cb, NULL);
}
```

In the push_btn_cb, you can turn on/off the light and also report the state change to HomeKit by calling `hap_char_update_val()`

matheusesperanca, 2021-10-02:
Solved, thanks!

closed 2021-10-02 by matheusesperanca

pwallner, 2021-10-17:
After hours of searching, I found out...
```
//turn on light
hap_val_t appliance_value = {
        .b = true
};
hap_char_t *temp = hap_serv_get_char_by_uuid(service, HAP_CHAR_UUID_ON);
hap_char_update_val(temp, &appliance_value);
```