espressif/esp-homekit-sdk #75: Error: "Accessory Already Added" in HomeKit

ankurag12, 2021-11-05:
I was adding and removing ESP32 accessory, trying out a few things in the code (starting from fan example) and it was running smoothly but once after removing the accessory, Homekit refused to add accessory saying that the accessory has already been added. I’ve tried

- Deleting the home/room
- Restarting iPhone
- Restarting, Reflashing the ESP32 device to another example code
- Changing ESP32 device name, serial number in accessory configuration
- Using another iPhone

Not sure if it's an ESP or HomeKit issue, but any help would be appreciated!
[image]

shahpiyushv, 2021-11-08:
@ankurag12 , for me, adding manually instead of via QR code had worked. Have you tried that?

ankurag12, 2021-11-08:
I tried adding manually (using code 11122333) but the app still isn't able to detect the accessory
[image]

ankurag12, 2021-11-09:
It seems like the device was thinking it was still paired, even though the HomeKit app showed otherwise. I was able to clean the pairing information using this command 
```
esptool.py -p $ESPPORT erase_region 0x10000 0x6000
```
and it works now!
Let me know if there's a better way to clean the pairing information.

shahpiyushv, 2021-11-09:
@ankurag12 , the cleanest way to remove an accessory is to delete it from Home app while it is reachable. Otherwise, you can press and hold the boot button for more than 10 seconds to trigger a factory reset.

The solution i mentioned earlier applies to cases wherein iOS throws an error even when accessory is removed from the Home app and is also factory reset.

wdqwd3213, 2022-04-23:
> It seems like the device was thinking it was still paired, even though the HomeKit app showed otherwise. I was able to clean the pairing information using this command
>
> ```
> esptool.py -p $ESPPORT [...]

hey, i have the same problem but im not really a tech guy... where can i type in the command?

shahpiyushv, 2022-04-25:
@wdqwd3213 if you are asking about `esptool.py -p $ESPPORT erase_region 0x10000 0x6000`, then it is executed from your host's terminal, from the same place where you execute `idf.py flash`

wdqwd3213, 2022-04-25:
no' i mean what is the host terminal? i didnt even know iphone had commands. i need the most basic explentions

shahpiyushv, 2022-04-26:
@wdqwd3213 , The "host" here is your development host, which you use for compilation and flashing. If you are using VS code or similar IDE, just erase the flash and re-flash the example using the commands mentioned [here](https://github.com/espressif/vscode-esp-idf-extension#available-commands).

Kenerick, 2023-04-03:
Is there a video for this specific issue?

shahpiyushv, 2023-04-07:
@Kenerick , not sure what video you are looking for here. Anyways, closing this issue since a solution was provided. Moreover [this commit](https://github.com/espressif/esp-homekit-sdk/commit/efca14e7bc9ca9c4fc08cc5ce846a8fe6435ed24) also fixes a related issue wherein iOS shows all accessories as "Blinds" when the QR code is scanned.

closed 2023-04-07 by shahpiyushv