espressif/esp-homekit-sdk #77: Wi-Fi compliance

smartsetup, 2021-11-30:
The last part to have my MFi homekit certification is to be a member of the Wi-Fi alliance. If I join, can I have my device certified as a derivative device without additional testing? What I asking is, does Espressif have a homekit Wi-Fi certified device that I can derive my accessory from? My current accessory is a homekit garage opener.

shahpiyushv, 2021-12-01:
@smartsetup , please check out [this blog](https://blog.espressif.com/wi-fi-certification-with-esp32-311e09dd06ff) to understand the derivative certification process. Since you are going to certification, I guess you are already using the MFi compliant variant of the SDK. This same pointer can also be found in the docs there.

smartsetup, 2021-12-01:
@shahpiyushv, thanks for that. It was really helpful. The article mentions ESP32-WROOM-32D as being already certified. I am using the ESP32-WROOM-32 module, is that certified too? Thanks

shahpiyushv, 2021-12-02:
@smartsetup , please refer [this page](https://www.espressif.com/en/support/documents/certificates?keys=) for information about various modules and their certifications. if you do not find the information you are looking for, please send an email to sales@espressif.com

shahpiyushv, 2022-09-27:
Closing this issue since it has been answered.

closed 2022-09-27 by shahpiyushv