espressif/esp-homekit-sdk #78: Wi-Fi re-configuration

smartsetup, 2021-12-06:
Hi, as part of MFi requirements it needs that Wi-Fi re-configuration service to be enabled. How do I go about doing that? Thanks

smartsetup, 2021-12-07:
I have just been informed by Apple that Espressif SDK is based on the HAP Spec R16, when is Espressif planning on release the new firmware that is based on HAP Spec R16? Also I have been told that they will only accept R16 so I am currently stuck. Help please. Thanks

shahpiyushv, 2021-12-07:
@smartsetup , the HAP R16 support has already available since long on the MFi compliant variant of the SDK on a separate branch. Please check that out.

smartsetup, 2021-12-07:
Thanks, the branch documentation says that it is not stable, have it passed all of the relevant test cases? Will it be OK to use in a commercial product? Thanks

shahpiyushv, 2021-12-07:
@smartsetup , it indeed passes all the new tests. If you see any failures for your specific use case, you can contact us on the MFi specific git repository or via sales@espressif.com

smartsetup, 2021-12-08:
I have got the R16 branch, the build fails. The folder below are empty:
components/homekit/json_generator/upstream/
components/homekit/json_parser/upstream/

once copied files from main repo, it compiles fine.

shahpiyushv, 2021-12-09:
@smartsetup As mentioned in the README, cloning using `git clone --recursive` or pulling the submodules using `git submodule update --init --recursive` would populate these folders. Meanwhile, closing this issue since the discussion here is not applicable for this GitHub repository.

closed 2021-12-09 by shahpiyushv