espressif/esp-homekit-sdk #79: Keepalive for stable performance

vvrec, 2021-12-14:
Hello everyone! If I understand correctly, this problem #18 is still relevant and I must use keepalive for stable performance. And one more question - this is also true for C3, S2, S3 ?

ifullgaz, 2021-12-15:
Hi there,

I am also seeing the `Decryption error/Connection lost` problem  https://github.com/espressif/esp-homekit-sdk/issues/18.

I'm using `esp-idf` 4.3.1 and the latest `esp-homekit-sdk`.
I've added some logs to be able to pinpoint the problem to some files:

`esp-idf/components/esp_http_server/httpd_parse.c`
`esp-homekit-sdk/components/homekit/esp_hap_core/src/esp_hap_pair_verify.c`
`esp-homekit-sdk/components/homekit/esp_hap_core/src/esp_hap_network_io.c`

Diff files are attached.

I tried with a custom board with an ESP32C3-MINI-1 and a standard dev board with an ESP32-WROOM-32.
I get the error with both boards at random period of times - sometimes after a few seconds, sometimes after many minutes.
The reason for disconnection is always that the socket receives 0 bytes. For instance (log from the ESP32C3):

```
D (156925) httpd: httpd_server: doing select maxfd+1 = 53
D (160395) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3fc9d2c4
D (160495) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3fc9d2c4
D (242275) httpd: httpd_server: processing socket 51
D (242275) httpd_sess: httpd_sess_process: httpd_req_new
D (242275) httpd_txrx: httpd_recv_with_opt: requested length = 128
Received 0 bytes from session 51
len = 0, session error
Decryption error/Connection lost. Marking session 51 as invalid
D (242295) event: no handlers have been registered for event HAP_EVENT:4 posted to loop 0x3fc9cfd0
D (242305) httpd_txrx: httpd_recv_with_opt: error in recv_fn
D (242305) httpd_parse: read_block: error in httpd_recv
D (242315) httpd_parse: httpd_parse_req: read_block() error fail
D (242325) httpd: httpd_server: closing socket 51
D (242325) httpd_sess: httpd_sess_delete: fd = 51
HomeKit Session 51 terminated
D (242335) httpd: httpd_server: doing select maxfd+1 = 53
D (242335) httpd: httpd_server: processing ctrl message
D (242345) httpd: httpd_process_ctrl_msg: work
D (242345) httpd_sess: httpd_sess_delete: fd = -1
D (242355) httpd: httpd_server: doing select maxfd+1 = 53
D (242685) httpd: httpd_server: processing socket 52
D (242685) httpd_sess: httpd_sess_process: httpd_req_new
D (242685) httpd_txrx: httpd_recv_with_opt: requested length = 128
Received 0 bytes from session 52
len = 0, session error
Decryption error/Connection lost. Marking session 52 as invalid
D (242705) event: no handlers have been registered for event HAP_EVENT:4 posted to loop 0x3fc9cfd0
D (242715) httpd_txrx: httpd_recv_with_opt: error in recv_fn
D (242715) httpd_parse: read_block: error in httpd_recv
D (242725) httpd_parse: httpd_parse_req: read_block() error fail
D (242725) httpd: httpd_server: closing socket 52
D (242735) httpd_sess: httpd_sess_delete: fd = 52
HomeKit Session 52 terminated
D (242745) httpd: httpd_server: doing select maxfd+1 = 50
D (242745) httpd: httpd_server: processing ctrl message
D (242755) httpd: httpd_process_ctrl_msg: work
D (242755) httpd_sess: httpd_sess_delete: fd = -1
D (242765) httpd: httpd_server: doing select maxfd+1 = 50
```

The `app_main.c` file is attached.

The problem seems more acute with the ESP32C3 than with the ESP32 - it seems to disconnect at random times.
Since I'm running the Home app on a Mac, the connection should remain opened - but it doesn't. Obviously, the device displays `No response` in the Home app.

I've tried everything, from setting the Keep alive option to disabling the GPIO that I want to use with ADC - just in case it was messing up with the WiFi signal. No luck.

Also, I've used WireShark to see if there was a corresponding disconnect TCP message from the Mac (Home app) when receiving 0 on the corresponding socket. There was literally no message sent by the Mac at the time of disconnection.

I can't find a good explanation for that behavior, I was hoping someone more knowledgable than me could help with finding answers and solutions.

Thanks
[esp-homekit-sdk.txt](https://github.c
[app_main.txt](https://github.com/espressif/esp-homekit-sdk/files/7720072/app_main.txt)
om/espressif/esp-homekit-sdk/files/7720064/esp-homekit-sdk.txt)
[esp-idf.txt](https://github.com/espressif/esp-homekit-sdk/files/7720065/esp-idf.txt)
!

vvrec, 2021-12-15:
I noticed one thing. I have several different smart home systems. One works with the budget Halo S12 router, the other works with the premium Ubiquiti AmpliFi. There is no problem with the AmpliFi system. But I don’t know much about networks to know how to configure Halo (I tried many different options for configuring the router.).

ifullgaz, 2021-12-16:
I've captured the disconnection packets using WireShark and included a file here.
It's in JSON format if anyone is interested in having a look.

I'm baffled by this problem.

[Disconnect.txt](https://github.com/espressif/esp-homekit-sdk/files/7730327/Disconnect.txt)

closed 2022-07-29 by vvrec