espressif/esp-homekit-sdk #80: How to include esp_hap_database.h file in app_main.c

twlawrenceko, 2021-12-29:
Hello,

As HAP uses port 80, we would like to redirect port 80 to port 8080 once the user hits port 80 (i.e. 192.168.1.1 --> 192.168.1.1:8080). However, the uri handler is in HAP and the `httpd_handle_t` is defined in [esp_hap_database.h](https://github.com/espressif/esp-homekit-sdk/blob/b3ca71eb1403d5c190f0e6b9ef6dbce228eff50d/components/homekit/esp_hap_core/src/priv_includes/esp_hap_database.h#L84). In order to create a redirect, a `httpd_register_uri_handler(hap_priv.server, &port_redirect_post_uri)` command is needed. However, even with `#include <esp_hap_database.h>` added on top of app_main.c it says `fatal error: esp_hap_database.h: No such file or directory`.

Please assist on how to configure CMakeLists file or others to be possible to get `hap_priv.server`.

Below is a chain of function steps:
1. [`hap_httpd_start()`](https://github.com/espressif/esp-homekit-sdk/blob/b3ca71eb1403d5c190f0e6b9ef6dbce228eff50d/components/homekit/esp_hap_core/src/esp_hap_main.c#L322) inside [`hap_start()`](https://github.com/espressif/esp-homekit-sdk/blob/b3ca71eb1403d5c190f0e6b9ef6dbce228eff50d/components/homekit/esp_hap_core/src/esp_hap_main.c#L295).
2. [`hap_platform_httpd_start(&hap_priv.server)`](https://github.com/espressif/esp-homekit-sdk/blob/c62f64dea6669547182e932dfded0a3a912a1951/components/homekit/esp_hap_core/src/esp_hap_ip_services.c#L1481) called inside [`hap_httpd_start()`](https://github.com/espressif/esp-homekit-sdk/blob/b3ca71eb1403d5c190f0e6b9ef6dbce228eff50d/components/homekit/esp_hap_core/src/esp_hap_main.c#L322).
3. [`hap_platform_httpd_start(httpd_handle_t *handle)`](https://github.com/espressif/esp-homekit-sdk/blob/b3ca71eb1403d5c190f0e6b9ef6dbce228eff50d/components/homekit/esp_hap_platform/src/hap_platform_httpd.c#L27) uses `&hap_priv.server` that is pointed into it.

Thank you

shahpiyushv, 2021-12-29:
@twlawrenceko , what exactly is the purpose of what you are doing? If you want to use port 8080 instead of 80 for HAP, you might as well change it using CONFIG_HAP_HTTP_SERVER_PORT

twlawrenceko, 2021-12-29:
> @twlawrenceko , what exactly is the purpose of what you are doing? If you want to use port 8080 instead of 80 for HAP, you might as well change it using CONFIG_HAP_HTTP_SERVER_PORT

I am not familiar with HAP protocols, would there be any issue or trouble if we use port 8080 for HAP instead of port 80? A http webpage is hosted but for the ease of use, we would like to enter the hostname and redirect to port 8080 automatically.

shahpiyushv, 2021-12-29:
@twlawrenceko , HAP can work on any port, just that it should match with what is announced in the HAP mDNS service. `CONFIG_HAP_HTTP_SERVER_PORT` takes care of that.

shahpiyushv, 2021-12-29:
@twlawrenceko , just to add, even if you want to use the hap_priv.server pointer to add any new URIs, instead of accessing the internal DB, you can use the [`hap_platform_httpd_get_handle()`](https://github.com/espressif/esp-homekit-sdk/blob/c62f64dea6669547182e932dfded0a3a912a1951/components/homekit/esp_hap_platform/include/hap_platform_httpd.h#L131) API, an example of which can be found [here](https://github.com/espressif/esp-homekit-sdk/blob/c62f64dea6669547182e932dfded0a3a912a1951/components/homekit/esp_hap_extras/src/hap_bct_http_handlers.c#L76).

twlawrenceko, 2021-12-30:
@shahpiyushv , Thank you for the detailed info. 
This is an extended question regarding to this: The `mdns_hostname_set()` does not seem to when HAP is running. Is there any method to fix this?

shahpiyushv, 2021-12-30:
@twlawrenceko do you mean to say that if you change the hostname using `mdns_hostname_set()`, it does not take effect? Please use it after `hap_start()` and see if it works.

twlawrenceko, 2022-01-10:
> @twlawrenceko do you mean to say that if you change the hostname using `mdns_hostname_set()`, it does not take effect? Please use it after `hap_start()` and see if it works.

Thank you for the notice, after `hap_start()` it works.

closed 2022-01-10 by twlawrenceko

twlawrenceko, 2022-12-06:
Hello @shahpiyushv ,

It seems to be related issue to this one which HAP is conflicting with the httpd_server connection.

We have the follow error when visiting REST server with HAP running at the same time:
```
W (50673) httpd: httpd_accept_conn: error in accept (23)
W (50673) httpd: httpd_server: error accepting new connection
```

Before merging HAP, we have enhanced the reliability in WS and REST server (based on [restful_server](https://github.com/espressif/esp-idf/tree/release/v4.4/examples/protocols/http_server/restful_server) and [wss_server](https://github.com/espressif/esp-idf/tree/release/v4.4/examples/protocols/https_server/wss_server)), `CONFIG_LWIP_SO_LINGER=y` is also set in menuconfig along with [`httpd.close_fn` modifications ](https://github.com/espressif/esp-idf/issues/9395).
And to merge with HAP gracefully with REST server, we have changed `CONFIG_HAP_HTTP_SERVER_PORT=8080` and `CONFIG_HAP_HTTP_MAX_OPEN_SOCKETS=7` in menuconfig for HAP to avoid port conflicts.

How can we fix the random `error in accept (23)` error in this case along with HAP enabled?