espressif/esp-homekit-sdk #82: Error building smart_plug example form esp-homekit-sdk

Sirius3615, 2022-01-21:
I get a build error every time i try to build smart_plug example without modifying anything. I set my board type right and the port right. Any clues how to fix this?

Terminal output: https://gist.github.com/Sirius3615/08263acbf11d2318565d7545258e7c0

Screenshot form VS code:
[image]

I am using MacBook Air M1, I can normally build and flash using regular esp-idf.

shahpiyushv, 2022-01-26:
@Sirius3615, what is your esp idf version? If it's master branch, it may or may not work. Can you try with release/v4.3 branch? The problem could very well be something different, but using the right branch would be worth a try :)

Sirius3615, 2022-01-26:
Hi, I figured it our with help form a friend, I had to enable Bluetooth 4.2 in menuconfig since I have esp32c3. After enabling that I was able to build and flash it. Only issue is that neither for me or my friend the QR code for provisioning doesn't work. It just doesn't wanna find the device. Thanks!