espressif/esp-homekit-sdk #83: Fail Target ESP32-S3

brunohorta82, 2022-02-02:
I try with ESP-IDF v4.4-dirty and ESP-IDF v5.0

to replicate execute the command:  idf.py set-target esp32s3

Result: 

-- Project sdkconfig file /Users/brunohorta/esp/esp-homekit-sdk/examples/fan/sdkconfig
Loading defaults file /Users/brunohorta/esp/esp-homekit-sdk/examples/fan/sdkconfig.defaults...
Traceback (most recent call last):
  File "/Users/brunohorta/AvanceeGit/corex_firmware/CoreX-IDF/main/idf_4_4/esp-idf/tools/kconfig_new/confgen.py", line 640, in <module>
    main()
  File "/Users/brunohorta/AvanceeGit/corex_firmware/CoreX-IDF/main/idf_4_4/esp-idf/tools/kconfig_new/confgen.py", line 326, in main
    output_function(deprecated_options, config, temp_file)
  File "/Users/brunohorta/AvanceeGit/corex_firmware/CoreX-IDF/main/idf_4_4/esp-idf/tools/kconfig_new/confgen.py", line 473, in write_json
    config_dict = get_json_values(config)
  File "/Users/brunohorta/AvanceeGit/corex_firmware/CoreX-IDF/main/idf_4_4/esp-idf/tools/kconfig_new/confgen.py", line 468, in get_json_values
    write_node(n)
  File "/Users/brunohorta/AvanceeGit/corex_firmware/CoreX-IDF/main/idf_4_4/esp-idf/tools/kconfig_new/confgen.py", line 465, in write_node
    val = int(val)
ValueError: invalid literal for int() with base 10: ''
CMake Error at /Users/brunohorta/AvanceeGit/corex_firmware/CoreX-IDF/main/idf_4_4/esp-idf/tools/cmake/kconfig.cmake:266 (message):
  Failed to run confgen.py
  (/Users/brunohorta/.espressif/python_env/idf4.4_py3.9_env/bin/python;/Users/brunohorta/AvanceeGit/corex_firmware/CoreX-IDF/main/idf_4_4/esp-idf/tools/kconfig_new/confgen.py;--kconfig;/Users/brunohorta/AvanceeGit/corex_firmware/CoreX-IDF/main/idf_4_4/esp-idf/Kconfig;--sdkconfig-rename;/Users/brunohorta/AvanceeGit/corex_firmware/CoreX-IDF/main/idf_4_4/esp-idf/sdkconfig.rename;--config;/Users/brunohorta/esp/esp-homekit-sdk/examples/fan/sdkconfig;--defaults;/Users/brunohorta/esp/esp-homekit-sdk/examples/fan/sdkconfig.defaults;--env-file;/Users/brunohorta/esp/esp-homekit-sdk/examples/fan/build/config.env).
  Error 1
Call Stack (most recent call first):
  /Users/brunohorta/AvanceeGit/corex_firmware/CoreX-IDF/main/idf_4_4/esp-idf/tools/cmake/build.cmake:528 (__kconfig_generate_config)
  /Users/brunohorta/AvanceeGit/corex_firmware/CoreX-IDF/main/idf_4_4/esp-idf/tools/cmake/project.cmake:378 (idf_build_process)
  CMakeLists.txt:17 (project)

-- Configuring incomplete, errors occurred!
See also "/Users/brunohorta/esp/esp-homekit-sdk/examples/fan/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1

shahpiyushv, 2022-03-02:
@brunohorta82 esp-homekit-sdk may not work with esp-idf master branch yet, but you can apply this attached patch in esp-homekit-sdk, so that it can compile for esp32s3 with idf v4.4

[esp32s3_patch.txt](https://github.com/espressif/esp-homekit-sdk/files/8171857/esp32s3_patch.txt)

shahpiyushv, 2022-03-25:
Support for ESP32S3 is now available.

closed 2022-03-25 by shahpiyushv