espressif/esp-homekit-sdk #84: Failed to resolve component 'libsodium'

heartbreaker0, 2022-02-14:
I get this error when I tried to build the fan example. I'm using esp-idf 4.4 under Ubuntu21.04 and I installed libsodium via apt and compilation.
[image]
 Thanks for your help.

closed 2022-02-15 by heartbreaker0

WallXiaoming, 2022-02-18:
I have the same problem. How you solve it? thanks

WallXiaoming, 2022-02-18:
> I get this error when I tried to build the fan example. I'm using esp-idf 4.4 under Ubuntu21.04 and I installed libsodium via apt and compilation. [image] Thanks for your help.

I have the same problem. How you solve it? thanks

Roelof626, 2022-03-06:
For me too, really want to solve it. already tried several things to build the project successfully...

shahpiyushv, 2022-03-07:
@Roelof626 @WallXiaoming , for the time being,  I would recommend that you use release/v4.4 (or an other release/v4.x) branch of esp-idf instead of the master branch. The master branch has moved on to 5.0, which may not be backward compatible with 4.x.