espressif/esp-homekit-sdk #91: How to implement automatic shutdown in switch module and update homekit state

ilovn, 2022-08-08:
I implemented a switch module through the code example, which controls the device through the gpio control output. Now I hope to automatically turn off 30 seconds after opening. I realized the shutdown through the timer, but I can't update the state in the homekit.

shahpiyushv, 2022-08-08:
@ilovn You can maintain a pointer of the characteristic and update it as shown in [this example](https://github.com/espressif/esp-homekit-sdk/blob/fac2032426d3cd29d8b6cc2663d0e7945d1d020d/examples/smart_outlet/main/app_main.c#L230). If you do not want to maintain the pointer, you can also fetch it at runtime using a combination of `hap_acc_get_by_aid()`, `hap_acc_get_serv_by_uuid()` and `hap_serv_get_char_by_uuid`, all of which can be found in [hap.h](https://github.com/espressif/esp-homekit-sdk/blob/master/components/homekit/esp_hap_core/include/hap.h)

ilovn, 2022-08-09:
When I use GPIO18 to control the relay to realize the circuit on and off, but updating the Homekit status still does not work properly, which UUID type with `hap_serv_get_char_by_uuid` or `hap_acc_get_serv_by_uuid` ? I got the aid from log,but `hap_acc_get_by_aid` still does not work.

shahpiyushv, 2022-08-09:
Since your implementation is a simple switch, you just need to call hap_acc_get_by_aid(1). This pointer is created and returned by hap_acc_create() and so you can keep a copy of that too. Standard service UUIDs can be found [here](https://github.com/espressif/esp-homekit-sdk/blob/master/components/homekit/esp_hap_apple_profiles/include/hap_apple_servs.h) and standard characteristic UUIDs can be found [here](https://github.com/espressif/esp-homekit-sdk/blob/master/components/homekit/esp_hap_apple_profiles/include/hap_apple_chars.h). UUID for switch is HAP_SERV_UUID_SWITCH, whereas that of outlet is HAP_SERV_UUID_SWITCH. The UUID for ON characteristic is HAP_CHAR_UUID_ON.

ilovn, 2022-08-10:
Through the UUID or aid you provide, the code can run normally, but the switch status in the Home App remains unchanged.

shahpiyushv, 2022-08-11:
Has iOS enabled the events? You can verify this by checking for prints like these in the serial terminal logs

```
Events enabled for aid=x iid=y
```

If you see this, whenver you call `hap_char_update_val()`, you should also see prints like

```
Value Changed
Notification Sent
```

You may also enable additional debugging using `hap_http_debug_enable()` once before calling `hap_start()` so that we can more logs to understand the issue.

NguyenVux, 2024-05-19:
mine print out `Events disabled for aid=x iid=y` how can i enable it

shahpiyushv, 2024-05-20:
Event subscription is managed by the iOS layer and is not in user's or accessory's control. If the serial terminal prints `Events disabled for aid=x iid=y` it means that iOS has explicitly disabled it since it is no more required. This typically happens when the Home app goes in background.