espressif/esp-homekit-sdk #92: Not possible to set fan rotation speed step value

ahharvey, 2022-08-09:
#### Overview

The SDK does not expose the step attribute of HAP's rotation speed characteristic. As a result, it os not possible to modify the step value to accommodate e.g., 3-speed fans. 

#### Detailed description

The Homekit specification indicates the fan service has a rotation speed characteristic. 

The specification also indicates that the rotation speed characteristic accepts two parameters: (i) fan speed - a float indicating percentage speed (0-100); and (ii) step - a float indicating the interval between fan speeds and used for configuring e.g., 3-speed fans (step = 33.3%). 

esp-homekit-sdk provides `hap_char_rotation_speed_create` to configure the rotation speed characteristic. However, this function does not expose an api to configure the step. 

#### Solution 

1. To be consistent with the HAP specification, esp-homekit-sdk should expose the step argument on hap_char_rotation_speed_create.
2. Any changes to hap_char_rotation_speed_create should be backward compatible and not break existing implementations.

ahharvey, 2022-08-09:
I have submitted a PR that fixes this issue, for consideration of the maintainers.

shahpiyushv, 2022-09-27:
Closing as per the explanation here: https://github.com/espressif/esp-homekit-sdk/pull/93#issuecomment-1209661254

closed 2022-09-27 by shahpiyushv