espressif/esp-homekit-sdk #94: Running console example alongside Homekit causes error

tobyloki, 2022-08-28:
I am trying to run the advanced console example from https://github.com/espressif/esp-idf/tree/master/examples/system/console/advanced alongside the Homekit sdk. I initialized the console application first and then Homekit. After calling hap_start(), I get the error HAP mDNS Init failed. How can I fix this problem?

shahpiyushv, 2022-08-29:
@tobyloki , can you try enabling debug logs using idf.py menuconfig -> Component config -> Log output -> Default log verbosity -> Debug and check the serial console prints?

tobyloki, 2022-08-29:
These are the logs leading up to the error:
```
Getting setup info from factory NVS
D (3460) nvs: nvs_open_from_partition hap_setup 0
D (3470) nvs: nvs_get_str_or_blob setup_salt
D (3470) nvs: nvs_close 39
D (3470) nvs: nvs_open_from_partition hap_setup 0
D (3470) nvs: nvs_get_str_or_blob setup_verifier
D (3490) nvs: nvs_close 40
D (3490) nvs: nvs_open_from_partition hap_main 0
D (3490) nvs: nvs_get_str_or_blob fw_rev
D (3490) nvs: nvs_close 41
HAP Main Loop Started
D (3580) httpd: httpd_thread: web server started
D (3580) httpd: httpd_server: doing select maxfd+1 = 50
D (3580) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3ffdd5dc
D (3580) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x0
D (3590) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x0
HAP mDNS Init failed
```

shahpiyushv, 2022-08-30:
@tobyloki are you also initialising Wi-Fi using `app_wifi_init()` like shown in HomeKit [examples](https://github.com/espressif/esp-homekit-sdk/blob/master/examples/smart_outlet/main/app_main.c#L209)?