espressif/esp-homekit-sdk #95: IDF v5 support

mbuckaway, 2022-09-13:
I'm wondering about support for v5.0 of the IDF. Currently, attempting to compile against this version of the IDF does not work.

There are a few minor changes that need to be made along with some changes to the build system. Warning as errors is enabled by default. For example:

hkdf.c:315:
int hkdfResult(HKDFContext *context,
               uint8_t prk[USHAMaxHashSize],
               const unsigned char *info, int info_len,
               uint8_t okm[USHAMaxHashSize], int okm_len)

okm requires a fully qualified type.

hmac.c:221L
int hmacResult(HMACContext *context, uint8_t digest[USHAMaxHashSize])

digest requires a fully qualified type.

As for the build system, I am unable to find the CMakefile combination that allows the code to compile.

It would appear the following is required:
 idf_component_register(SRC_DIRS src
        REQUIRES esp_hw_support mbedtls log driver nvs_flash esp_event esp_wifi esp32-button app_hap_setup_payload esp_hap_apple_profiles esp_hap_core esp_hap_platform esp_system
        INCLUDE_DIRS include)

...but each attempt gets a different result of a missing routine.

Any recommendations?

shahpiyushv, 2022-09-13:
@mbuckaway , the release v5.0 is not yet tagged and there have been quite some breaking changes, some of which are still ongoing. Once the changes are frozen internally, we will add support for it

loicmorvan, 2023-01-30:
As you say in #99 , [esp-idf v5.0 is now out](https://github.com/espressif/esp-idf/tree/v5.0).

I tried building the `lightbulb` example with that version of idk, and here are my two cents:
- the component `mdns` is no more part of IDF and should be added as dependency from the idk component registry `idf.py add-dependency mdns`.
- Some sources should be updated because the signature of some functions switched from `uint8_t *` or `uint8 []` to `uint8_t [USHAMaxHashSize]`, in `esp-homekit-sdk/components/homekit/hkdf-sha/upstream/hkdf.c:318:24` and in `esp-homekit-sdk/components/homekit/hkdf-sha/upstream/hmac.c:221:47`.
- In `esp-homekit-sdk/components/homekit/esp_hap_core/src/esp_hap_acc.c:25:10`, `esp_wifi.h: No such file or directory`, that one I don't know how to resolve :)

mbuckaway, 2023-01-30:
@loicmorvan , most of the missing head file issues are result of Espressif removing the default modules from the "regular" config. You will have to spend many compile cycles to figure out the dependancies and add them to the CMakefile (ie. esp_wifi, and some 20 others). Check the release notes for IDF 5 for more details. I eventually gave up and just used IDF 4.4 because I didn't need any of the new features. I suspect Espressif will fix the Homekit module eventually.

loicmorvan, 2023-01-30:
@mbuckaway , thanks for your post, I saw that you have tried that a few months ago. I don't master *cmake* (yet!) nor the framework IDF, but I will try making HomeKit working with IDF 5. I'll keep you posted and maybe try to launch a Pull Request.

shahpiyushv, 2023-02-06:
Sorry for the delay. We will check this on priority.

shahpiyushv, 2023-03-01:
Support for idf v5.0 has been added now.