espressif/esp-homekit-sdk #96: esp-homekit-sdk API使用问题

kuangswxizhenelec, 2022-09-26:
我们的项目应用为homekit bridge配件，桥接的配件数量可能会根据需要增加或者删减，此外相关桥接配件的服务或特征也有可能动态改变，同时Home APP内对应的图标也要求隐藏或显示。相应的对此我有以下几点疑问：

上述行为（动态增删已桥接的配件以及动态修改服务和特征）是否被允许且符合homekit 配件协议规范（R17 or R16）？

Home APP内的图标（比如温度调节和转速调节）是否可以被隐藏或显示？

如果允许如上述所说动态更改，那需要怎样操作来动态增加或删除配件，以及配件的对应服务和特征？
实际上我在Hap.h内找到几个API如下 以上API都与“删除”相关，删除之后Home APP显示“未响应”而不是隐藏图标;而且我不知道删除之后如何再恢复该配件或服务或特征，尝试了很多但Home App内配件还是显示未响应。void hap_remove_bridged_accessory(hap_acc_t *ha);void hap_acc_delete(hap_acc_t *ha);void hap_delete_all_accessories(void);void hap_char_delete(hap_char_t *hc);void hap_serv_delete(hap_serv_t *hs);

void hap_serv_mark_hidden(hap_serv_t *hs);这个API我使用后没有任何改变，我不明白是什么意思，我需要如何使用它？

shahpiyushv, 2022-09-27:
@kuangswxizhenelec , using `hap_add_bridged_accessory()` or `hap_remove_bridged_accessory()` will trigger a config number (c#) update on mDNS. Based on this iOS should re-fetch all the accessory information. I would recommend that you first try only this to see how it goes. You can verify the c# updates by checking for it in the _hap mDNS announcement seen in [Bonjour Discovery apps](https://apps.apple.com/us/app/discovery-dns-sd-browser/id1381004916?mt=12)

`hap_delete_all_accessories()` is not to be used normally as it deletes the primary accessory too. We might even remove this API altogether, so that it does not get misused.

What exactly is the use case requiring dynamic service/characteristic addition/removal? We do not update the config number for these operations, but you can do so yourself by calling `hap_update_config_number()` after you are done with your changes to the database.

kuangswxizhenelec, 2022-09-28:
Thanks, now I know how to add and remove accessories.

My bridged device uses both "thermostat" and "fan" services and allows one of them to stop working during use. At this point we want the Home APP to hide the UI of the non-working service instead of showing "unresponsive" all the time. I tried calling hap_update_config_number() to update the changes, but it seemed to cause ESP32 to reset or the entire bridge to be unresponsive.