espressif/esp-homekit-sdk #99: ESP IDF v5.0 support

vihanb, 2022-12-25:
I was attempting to run the example "fan" project. I notice the Makefile has line `include $(IDF_PATH)/make/project.mk`. This fails as I am using ESP IDF v5.0 which does not seem to have a `$IDF_PATH/make/` directory. Should my v5.0 installation work? Is there a suggested workaround for working with ESP IDF v5.0?

vihanb, 2022-12-25:
Looks like this is a duplicate of #95

closed 2022-12-25 by vihanb

shahpiyushv, 2022-12-27:
@vihanb ESP IDF v5.0 does not support "make", but uses  cmake instead. So the file to check is CMakeLists.txt, which would be same for v4.x as well as v5.x. However, there could be other compilation errors as v5.0 has some breaking changes. We will fix those soon since idf v5.0 has now been officially released.