espressif/esp-homekit-sdk #100: Custom services and characteristics do not show up in the Home app

Brawrdon, 2023-01-25:
I took a look at the [data_tlv8](https://github.com/espressif/esp-homekit-sdk/tree/eade505ade6c148056044307ce7b779bbc8e218b/examples/data_tlv8) example to see how I could add custom services and characteristics, the example connects to HomeKit fine, but only the temperature service is visible within the app.

I've played around as I've been trying to create a custom service that would have an Apple characteristic but I've had no luck so far. When I include a custom service/characteristic I notice the `Decryption error/Connection lost. Marking session as invalid` error appears more but the connection seems to recovery quickly.

Thanks for any help 😁

shahpiyushv, 2023-01-27:
As far as I know, the Home app does not show any custom services or characteristic. Moreover, if it finds anything invalid in the accessory database, it closes the connection. The database being reported in JSON can be seen if you enable http debugging by calling  `hap_http_debug_enable()` before `hap_start()`.

Brawrdon, 2023-01-27:
Thanks, I'll give that a go later and report if I spot anything weird.

What's the point of having a custom services or characteristic then, if they can't be seen in the app to do things like create automations? I assume the only use would be in apps that integrate with the HomeKit API to control the device?

shahpiyushv, 2023-01-27:
@Brawrdon , that is right. Manufacturers have their own HomeKit compatible apps to make use of the custom services/characteristics. Firmware developers typically use the HomeKit Accessory Tester (HAT) for testing this. However, I suppose this tool is available only to Apple MFI licensees.

Brawrdon, 2023-01-29:
Tried using `hap_http_debug_enable()` and it seems my custom characteristic is appearing! Thanks for the clarification 💗

closed 2023-01-29 by Brawrdon