espressif/esp-homekit-sdk #101: IDF V5.0 Results is Lexing error

ginodecock, 2023-01-27:
Hi,

When building the smart_outlet example in IDF V5.0 results in a Lexing error.
Any idea what this error causes?

> idf-v5.0/components/protocomm Y:/b/esp-idf-v5.0/components/pthread Y:/b/esp-homekit-sdk/examples/common/qrcode Y:/b/esp-idf-v5.0/components/sdmmc Y:/b/esp-idf-v5.0/components/soc Y:/b/esp-idf-v5.0/components/spi_flash Y:/b/esp-idf-v5.0/components/spiffs Y:/b/esp-idf-v5.0/components/tcp_transport Y:/b/esp-idf-v5.0/components/ulp Y:/b/esp-idf-v5.0/components/unity Y:/b/esp-idf-v5.0/components/usb Y:/b/esp-idf-v5.0/components/vfs Y:/b/esp-idf-v5.0/components/wear_levelling Y:/b/esp-idf-v5.0/components/wifi_provisioning Y:/b/esp-idf-v5.0/components/wpa_supplicant Y:/b/esp-idf-v5.0/components/xtensa
-- Configuring done
-- Generating done
CMake Error:
  Running

   'Y:/VM/Share/.esptools/tools/ninja/1.10.2/ninja.exe' '-C' 'Y:/b/esp-homekit-sdk/examples/smart_outlet/build' '-t' 'recompact'

  failed with:

   ninja: error: build.ninja:20474: **lexing error**

CMake Generate step failed.  Build files cannot be regenerated correctly.
cmake failed with exit code 1, output of the command is in the y:\b\esp-homekit-sdk\examples\smart_outlet\build\log\idf_py_stderr_output_22072 and y:\b\esp-homekit-sdk\examples\smart_outlet\build\log\idf_py_stdout_output_22072

loicmorvan, 2023-01-30:
This repository is not yet compatible with IDF v5.0. You can have a look at #95 .

shahpiyushv, 2023-03-01:
Support for idf v5.0 has been added now.