espressif/esp-homekit-sdk #108: Help me understand how homekit works (bridge + thermostat example)

MirekVer, 2023-08-11:
Hello everyone,
by studying the examples I was able to pairing with a bridge and add 10 thermostats.
I mainly have 3 problems/doubts that I can't solve:

1) I managed to implement the callback that triggers when the thermostat's current temperature, mode (inactive, hot, etc...), target temperature, etc... is changed.
However, this callback does not trigger if I change the name to the thermostat from the home app.
Surely I am doing something wrong, maybe I am considering the wrong characteristic 
`strcmp(hap_char_get_type_uuid(write->hc), HAP_CHAR_UUID_NAME) `
or something like that.

2) How can I change the name of a thermostat from code?
To read the current name of a thermostat, given the aid (thermo_aid):
```
hap_acc_t *ha = hap_acc_get_by_aid(thermo_aid);
hap_serv_t *hs = hap_acc_get_serv_by_uuid(ha, HAP_SERV_UUID_ACCESSORY_INFORMATION);
hap_char_t *hc = hap_serv_get_char_by_uuid(hs, HAP_CHAR_UID_NAME);
hap_val_t *val_read = (hap_val_t *)hap_char_get_val(hc);
ESP_LOGI(TAG, "Actual thermo name: %s", val_read->s);
```
I get the thermostat name correctly, so I modify it as follows:
```
hap_val_t val_write;
val_write.s = (char *)malloc(14);
memcpy(val_write.s, "new_name_test", 13);
ESP_LOGI(TAG, "New thermo name: %s", (val_write.s));
int ret = hap_char_update_val(hc, &val_write);
free(val_write.s);
```
(I also tried the same implementation by declaring val_write as a pointer).
If I run the code again to read the thermostat name, it returns the new changed name but on the home app I still see the old name and it is not updated.
What am I doing wrong?

3)Suppose everything works properly with 1 bridge and 10 thermostats.
I need to add two new thermostats (so from 10 thermostats -> 12 thermostats).
Can I do this while the hap server is running or should I stop and restart everything?
If I have to stop with hap_stop, do I have to redo the whole procedure after that command: hap init, bridge creation, 12 thermostats creation, etc.? 
If there are examples to do this I would be happy to study them but I can't find anything.

I hope someone can help me.

shahpiyushv, 2023-08-11:
@MirekVer , the name characteristic in the Accessory Information Service is a read-only characteristic and is used only for the default name. 
[image]
iOS does not send any updated values to the accessory but instead, maintains it only in its own database.

The HomeKit specs explicitly say that _The values of "Manufacturer", "Model", "Name", "Serial Number" "Product Data"  must be persistent through the lifetime of the accessory._, So you should not change the value as you have tried in the Q.2.

Any accessory can be added/removed at runtime using `hap_add_bridged_accessory()` and `hap_remove_bridged_accessory()`. You need not stop the hap framework for that.

MirekVer, 2023-08-11:
> @MirekVer , the name characteristic in the Accessory Information Service is a read-only characteristic and is used only for the default name. [image] iOS does not send any updated values to the [...]

Ok, thank you for the answer!

MirekVer, 2023-08-21:
No one can help me? :(

MirekVer, 2023-08-21:
> As you can see [here](https://github.com/espressif/esp-homekit-sdk/blob/3e6955ac2118f98b374b3f786776fe73f04daf1e/examples/bridge/main/app_main.c#L207) in the bridge example, we have assigned the accessory name as the private data. You can assign any data you want here and that data would be available as the `serv_priv` in the [write callback](https://github.com/espressif/esp-homekit-sdk/blob/3e6955ac2118f98b374b3f786776fe73f04daf1e/examples/bridge/main/app_main.c#L125). You can see that we have printed the same accessory name [here](https://github.com/espressif/esp-homekit-sdk/blob/3e6955ac2118f98b374b3f786776fe73f04daf1e/examples/bridge/main/app_main.c#L128).

Okay, it was much simpler than I thought.
Thx!