espressif/esp-homekit-sdk #110: Errors with example "lightbulb"

leotordo, 2023-09-18:
Hello,
I'm trying to compiler "lightbulb" example.
I use IDF 5.0.2 and VS Code
when I run "idy.py build" the terminal shows these errors

-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.0-2/components/soc/esp32/ld/esp32.peripherals.ld
CMake Error at C:/Espressif/frameworks/esp-idf-v5.0-2/tools/cmake/component.cmake:312 (message):
Include directory
'C:/esp-homekit-sdk-master/components/homekit/json_parser/upstream/include'
is not a directory.
Call Stack (most recent call first):
C:/Espressif/frameworks/esp-idf-v5.0-2/tools/cmake/component.cmake:486 (__component_add_include_dirs)
C:/esp-homekit-sdk-master/components/homekit/json_parser/CMakeLists.txt:1 (idf_component_register)
-- Configuring incomplete, errors occurred!
See also "C:/esp-homekit-sdk-master/examples/lightbulb/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1, output of the command is in the c:\esp-homekit-sdk-master\examples\lightbulb\build\log\idf_py_stderr_output_30000 and c:\esp-homekit-sdk-master\examples\lightbulb\build\log\idf_py_stdout_output_30000

Any idea? If you need some other informations like logs file, please ask me
Thanks

--update--

For error
C:/esp-homekit-sdk-master/components/homekit/json_parser/upstream/include' is not a directory.
I manually created it

now the error is

-- Configuring done
CMake Error at C:/Espressif/frameworks/esp-idf-v5.0-2/tools/cmake/component.cmake:484 (add_library):
Cannot find source file:
C:/esp-homekit-sdk-master/components/homekit/json_generator/upstream/json_generator.c
Tried extensions .c .C .c++ .cc .cpp .cxx .cu .mpp .m .M .mm .ixx .cppm .h
.hh .h++ .hm .hpp .hxx .in .txx .f .F .for .f77 .f90 .f95 .f03 .hip .ispc
Call Stack (most recent call first):
C:/esp-homekit-sdk-master/components/homekit/json_generator/CMakeLists.txt:1 (idf_component_register)
CMake Error at C:/Espressif/frameworks/esp-idf-v5.0-2/tools/cmake/component.cmake:484 (add_library):
Cannot find source file:
C:/esp-homekit-sdk-master/components/homekit/json_parser/upstream/src/json_parser.c
Tried extensions .c .C .c++ .cc .cpp .cxx .cu .mpp .m .M .mm .ixx .cppm .h
.hh .h++ .hm .hpp .hxx .in .txx .f .F .for .f77 .f90 .f95 .f03 .hip .ispc
Call Stack (most recent call first):
C:/esp-homekit-sdk-master/components/homekit/json_parser/CMakeLists.txt:1 (idf_component_register)
CMake Error at C:/Espressif/frameworks/esp-idf-v5.0-2/tools/cmake/component.cmake:484 (add_library):
No SOURCES given to target: __idf_json_generator
Call Stack (most recent call first):
C:/esp-homekit-sdk-master/components/homekit/json_generator/CMakeLists.txt:1 (idf_component_register)
CMake Error at C:/Espressif/frameworks/esp-idf-v5.0-2/tools/cmake/component.cmake:484 (add_library):
No SOURCES given to target: __idf_json_parser
Call Stack (most recent call first):
C:/esp-homekit-sdk-master/components/homekit/json_parser/CMakeLists.txt:1 (idf_component_register)
CMake Generate step failed.  Build files cannot be regenerated correctly.
cmake failed with exit code 1, output of the command is in the c:\esp-homekit-sdk-master\examples\lightbulb\build\log\idf_py_stderr_output_26516 and c:\esp-homekit-sdk-master\examples\lightbulb\build\log\idf_py_stdout_output_26516

Anyone can help me?
Regards

shahpiyushv, 2023-09-18:
@leotordo have you cloned the repository correctly including all submodules as mentioned [here](https://github.com/espressif/esp-homekit-sdk#get-esp-homekit-sdk)?

leotordo, 2023-09-19:
Hello,
 I did these steps
1) launched windows installer from Espressif website (windows version)
2) successfully built "blink" example of esp-idf-v5.1.1 framework from terminal
3) installed and configured VSCode
4) successfully built "blink" example of esp-idf-v5.1.1 framework from VSCode

After into dir of lightbulb example I launched command

git clone --recursive https://github.com/espressif/esp-homekit-sdk.git

and obtained this output

C:\esp-homekit-sdk-master\examples\lightbulb>git clone --recursive https://github.com/espressif/esp-homekit-sdk.git
Cloning into 'esp-homekit-sdk'...
remote: Enumerating objects: 698, done.
remote: Counting objects: 100% (196/196), done.
remote: Compressing objects: 100% (85/85), done.
remote: Total 698 (delta 122), reused 140 (delta 111), pack-reused 502
Receiving objects: 100% (698/698), 498.55 KiB | 1.83 MiB/s, done.
Resolving deltas: 100% (350/350), done.
Submodule 'components/homekit/json_generator/upstream' (https://github.com/espressif/json_generator.git) registered for path 'components/homekit/json_generator/upstream'
Submodule 'components/homekit/json_parser/upstream' (https://github.com/espressif/json_parser.git) registered for path 'components/homekit/json_parser/upstream'
Cloning into 'C:/esp-homekit-sdk-master/examples/lightbulb/esp-homekit-sdk/components/homekit/json_generator/upstream'...
remote: Enumerating objects: 53, done.
remote: Counting objects: 100% (8/8), done.
remote: Compressing objects: 100% (5/5), done.
remote: Total 53 (delta 3), reused 8 (delta 3), pack-reused 45
Receiving objects: 100% (53/53), 22.44 KiB | 3.74 MiB/s, done.
Resolving deltas: 100% (26/26), done.
Cloning into 'C:/esp-homekit-sdk-master/examples/lightbulb/esp-homekit-sdk/components/homekit/json_parser/upstream'...
remote: Enumerating objects: 51, done.
remote: Counting objects: 100% (26/26), done.
remote: Compressing objects: 100% (13/13), done.
remote: Total 51 (delta 7), reused 19 (delta 6), pack-reused 25
Receiving objects: 100% (51/51), 20.81 KiB | 3.47 MiB/s, done.
Resolving deltas: 100% (13/13), done.
Submodule path 'components/homekit/json_generator/upstream': checked out 'f508fc2eb29b3790a9871e493934956f133d8a8f'
Submodule path 'components/homekit/json_parser/upstream': checked out 'd049e7103f667c53a45bee5d27b7c15cb13b48c7'
Submodule 'jsmn' (https://github.com/zserge/jsmn.git) registered for path 'components/homekit/json_parser/upstream/jsmn'
Cloning into 'C:/esp-homekit-sdk-master/examples/lightbulb/esp-homekit-sdk/components/homekit/json_parser/upstream/jsmn'...
remote: Enumerating objects: 648, done.
remote: Counting objects: 100% (78/78), done.
remote: Compressing objects: 100% (42/42), done.
remote: Total 648 (delta 41), reused 64 (delta 36), pack-reused 570
Receiving objects: 100% (648/648), 276.05 KiB | 3.63 MiB/s, done.
Resolving deltas: 100% (319/319), done.
Submodule path 'components/homekit/json_parser/upstream/jsmn': checked out '053d3cd29200edb1bfd181d917d140c16c1f8834'

After I launched

C:\esp-homekit-sdk-master\examples\lightbulb>idf.py build

and get this

Executing action: all (aliases: build)
Running cmake in directory C:\esp-homekit-sdk-master\examples\lightbulb\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=C:\Espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe -DESP_PLATFORM=1 -DCCACHE_ENABLE=1 C:\esp-homekit-sdk-master\examples\lightbulb"...
-- IDF_TARGET is not set, guessed 'esp32' from sdkconfig 'C:/esp-homekit-sdk-master/examples/lightbulb/sdkconfig'
-- Found Git: C:/Espressif/tools/idf-git/2.39.2/cmd/git.exe (found version "2.39.2.windows.1")
-- Component directory C:/esp-homekit-sdk-master/components/homekit does not contain a CMakeLists.txt file. No component will be added
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe - skipped
-- Detecting C compile features
[... 29 lines cut ...]

  CMake Error at
  C:/esp-homekit-sdk-master/examples/common/app_wifi/CMakeLists.txt:3
  (project):

    project command is not scriptable

  Call Stack (most recent call first):

    C:/Espressif/frameworks/esp-idf-v5.1.1/tools/cmake/scripts/component_get_requirements.cmake:106 (include)
    C:/Espressif/frameworks/esp-idf-v5.1.1/tools/cmake/scripts/component_get_requirements.cmake:124 (__component_get_requirements)

Call Stack (most recent call first):
  C:/Espressif/frameworks/esp-idf-v5.1.1/tools/cmake/build.cmake:574 (__component_get_requirements)
  C:/Espressif/frameworks/esp-idf-v5.1.1/tools/cmake/project.cmake:547 (idf_build_process)
  CMakeLists.txt:17 (project)

-- Configuring incomplete, errors occurred!
See also "C:/esp-homekit-sdk-master/examples/lightbulb/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1, output of the command is in the C:\esp-homekit-sdk-master\examples\lightbulb\build\log\idf_py_stderr_output_34752 and C:\esp-homekit-sdk-master\examples\lightbulb\build\log\idf_py_stdout_output_34752

C:\esp-homekit-sdk-master\examples\lightbulb>

Whats wrong?
Thanks

shahpiyushv, 2023-09-20:
From your logs, it seems that the esp-homekit-sdk-master could be a direct zip download whereas the git repository is downloaded at `C:/esp-homekit-sdk-master/examples/lightbulb/esp-homekit-sdk/`. So, the example at `C:/esp-homekit-sdk-master/examples/lightbulb/esp-homekit-sdk/examples/lightbulb` is expected to work. However, to keep things clean, I would recommend to delete `C:/esp-homekit-sdk-master/` and clone `esp-homekit-sdk` directly in `C:/`.

leotordo, 2023-09-21:
solved. Thank you very much

closed 2023-09-21 by leotordo