espressif/esp-homekit-sdk #111: How send the led status to the iPhone?

leotordo, 2023-09-28:
Hello everybody.
I've got esp-homekit-sdk from Espressif and I built the example "lightbulb".
It works fine. I can change led status from may Home App.

Now, I want to switch on/off the led also by a mechanical switch.
I modified the code for reading a GPIO input connected to this switch.
I works, but the iPhone doesn't sense the status changed

How to send the changed status notification to the main task?

Thanks
Andrea

shahpiyushv, 2023-09-28:
You have to call [hap_char_update_val()](https://github.com/espressif/esp-homekit-sdk/blob/3e6955ac2118f98b374b3f786776fe73f04daf1e/components/homekit/esp_hap_core/include/hap.h#L1047) API to update the characteristic's value and also report it to the HomeKit controllers.

You can find sample usage in the smart_outlet example [here](https://github.com/espressif/esp-homekit-sdk/blob/3e6955ac2118f98b374b3f786776fe73f04daf1e/examples/smart_outlet/main/app_main.c#L232).

leotordo, 2023-09-28:
Thank you. I'll try ASAP and I'll turn here whit results
regards

leotordo, 2023-09-29:
Hello,
 I wrote this code
`
    while(1)
    {
            if(!gpio_get_level(GPIO_NUM_9)) 
            {
                lightbulb_set_brightness(50);
                
                hap_val_t appliance_value = { .i = 50, };
                hap_char_t *outlet_in_use = hap_serv_get_char_by_uuid(service, HAP_CHAR_UUID_BRIGHTNESS);
                hap_char_update_val(outlet_in_use, &appliance_value);
            }
    }
`
The application starts with 10% of PWM
`ret |= hap_serv_add_char(service, hap_char_brightness_create(10)); `

if I press GPIO9 switch the led brightness changes but the Home app remains at 10%

Where's the problem?
Thanks
Andrea

leotordo, 2023-09-29:
Brief update.
The position of my code is at the end of the thread _lightbulb_thread_entry_
I added a _single shot code_ for the button

    /* Enable Hardware MFi authentication (applicable only for MFi variant of SDK) */
    hap_enable_mfi_auth(HAP_MFI_AUTH_HW);

    /* Initialize Wi-Fi */
    app_wifi_init();

    /* After all the initializations are done, start the HAP core */
    hap_start();
    /* Start Wi-Fi */
    app_wifi_start(portMAX_DELAY);

    hap_val_t appliance_value = { .i = 50, };
    hap_char_t *outlet_in_use = hap_serv_get_char_by_uuid(service, HAP_CHAR_UUID_BRIGHTNESS);
    int first=0;
    while(1)
    {
            if(!gpio_get_level(GPIO_NUM_9) && first==0) 
            {
                first=1;
                lightbulb_set_brightness(50);
                              
                hap_char_update_val(outlet_in_use, &appliance_value);
            }
            if(gpio_get_level(GPIO_NUM_9) && first==1)
            {
                first=0;
                vTaskDelay(pdMS_TO_TICKS(300));
            }
    }
    /* The task ends here. The read/write callbacks will be invoked by the HAP Framework */
    vTaskDelete(NULL);
light_err:
    hap_acc_delete(accessory);
    vTaskDelete(NULL);
 }

and...doesn't work :-(
Thanks
Andrea

shahpiyushv, 2023-09-29:
@leotordo can you try adding prints to see if your code is reaching those lines? Additionally, it will be better to have a vTaskDelay(300/portTICK_PERIOD_MS); in your while(1) loop, outside the if conditions to prevent this task from consuming resources continuously.

leotordo, 2023-09-29:
lightbulb_set_brightness has prints
I added a print after hap_char_update_val (see below)

On the terminal I've these logs

Events Enabled for aid=1 iid=15
Events Enabled for aid=1 iid=17
Events Enabled for aid=1 iid=19
Events Enabled for aid=1 iid=20
Events Enabled for aid=1 iid=21

If I change from iPhone  I've this output

[image]

if I press mechanical switch I get

[image]

"Value Changed" doesn't appear....and the iPhone is not updated

```
while(1)
    {
           vTaskDelay(pdMS_TO_TICKS(300));
            if(!gpio_get_level(GPIO_NUM_9) && first==0) 
            {
                first=1;
                lightbulb_set_brightness(50);
                appliance_value.i=50;             
                hap_char_update_val(outlet_in_use, &appliance_value);
                ESP_LOGI(TAG, "hap_char_update_val SENT");
            }
            if(gpio_get_level(GPIO_NUM_9) && first==1)
            {
                first=0;
                vTaskDelay(pdMS_TO_TICKS(300));
            }
    }
```

leotordo, 2023-09-29:
I added the print of the returned value of hap_char_update_val

[image]

the error is **-1**

shahpiyushv, 2023-09-29:
A couple of points

- Check if `hap_serv_get_char_by_uuid(service, HAP_CHAR_UUID_BRIGHTNESS)` is indeed giving you a valid pointer. If you are using the lightbulb example directly, you will see [here](https://github.com/espressif/esp-homekit-sdk/blob/3e6955ac2118f98b374b3f786776fe73f04daf1e/examples/lightbulb/main/app_main.c#L218) that the service pointer gets over-riden with a different service:

```
    service = hap_serv_fw_upgrade_create(&ota_config);
    if (!service) {
        ESP_LOGE(TAG, "Failed to create Firmware Upgrade Service");
        goto light_err;
    }
    hap_acc_add_serv(accessory, service);
```

You can remove this snippet or use a different service pointer here.

- Use different values each time you call `hap_char_update_val(outlet_in_use, &appliance_value)`. If the value stays the same, no notification will be triggered.

leotordo, 2023-09-30:
thank you.
On monday I'll check your points.
Have a nice w/end

leotordo, 2023-10-02:
hello.
Issue solved.
You're right, the problem was on
**service = hap_serv_fw_upgrade_create(&ota_config);**

I defined service2 instead of service and my routine works.

I'll keep in mind also your suggestion about using different values.

Thanks
I close this issue
Regards

closed 2023-10-02 by leotordo