espressif/esp-homekit-sdk #113: Parameters accessory: what is ".pv" param?

leotordo, 2023-10-03:
In parameter accessory what means ".pv"?
"product version"?

Thanks   

  /* Initialise the mandatory parameters for Accessory which will be added as
     * the mandatory services internally
     */
    hap_acc_cfg_t cfg = {
        .name = "Esp-Light",
        .manufacturer = "Espressif",
        .model = "EspLight01",
        .serial_num = "abcdefg",
        .fw_rev = "0.9.0",
        .hw_rev = "1.0",
        .pv = "1.1.0",
        .identify_routine = light_identify,
        .cid = HAP_CID_LIGHTING,

shahpiyushv, 2023-10-04:
@leotordo , it is protocol version and is not supposed to be changed for regular use cases. This was retained just to support backward compatibility with protocol version v1.0.

leotordo, 2023-10-04:
thank you
regards

closed 2023-10-04 by leotordo