espressif/esp-homekit-sdk #115: Unable to Add Accessory - This accessory cannot be used with HomeKit

AMoo-Miki, 2023-10-11:
I pulled in the latest commits from the `release/v3.4` branch of [ESP8266_RTOS_SDK](https://github.com/espressif/ESP8266_RTOS_SDK) and from the `master` branch of this repository. My heads are at `fec8e98a` and `3e6955a` respectively.

I rebuilt and flashed a Garage Door Opener on an ESP8266, with code that has been working fine for months. I accidentally did an `erase_flash` and had to remove the accessory from Home to add it back.

Sadly, Home told me that it was `Unable to Add Accessory` and that `This accessory cannot be used with HomeKit`.

[image]

----

However, the logs on the ESP8266 says otherwise:
```
######## Starting Pair Setup ########
Pair Setup M1 Received
Pair Setup M2 Successful
I (63246) HAP-GarageDoor: Pairing Started
Pair Setup M3 Received
Using pair-setup without MFi.
Pair Setup M4 Successful
Pair Setup M5 Received
Pair Setup Successful for F4BD5D88-9094-4E49-B869-5E7872DD5759
I (71393) HAP-GarageDoor: Controller F4BD5D88-9094-4E49-B869-5E7872DD5759 Paired. Controller count: 1
Skipping s# update for Certification requirements.
Re-announcing _hap._tcp mDNS service
Cleaning Pair Setup Context
Cleaning Pair Setup Context
```

Rebooting the accessory, it says:
```
Accessory is Paired with atleast one controller
Updated state number to 2
Database initialised. Accessory Device ID: 50:9E:7C:F3:99:42
HAP Initialization succeeded. Version : 4.0-3e6955a
I (322) HAP-GarageDoor: Accessory is paired with 1 controllers
```

---

If I try to add the accessory to Home again, it says the "Accessory already in Another Home" which makes no sense because I have no other home!

[image]

---

I have erased flash, rebuilt, and reflashed numerous times, changed the Setup ID and Setup Code, and changed WiFi provisioning but nothing changes.

shahpiyushv, 2023-10-11:
Can you please call `hap_http_debug_enable()` before `hap_start()` and collect and share logs after that?

AramVartanyan, 2023-10-11:
@AMoo-Miki I am not sure what could cause the issue, but I had similar issue in the past.
I found that the solution for me was to set:

CONFIG_NEWLIB_NANO_FORMAT=n

in menuconfig.

Regards

AMoo-Miki, 2023-10-11:
Thanks for replying.

With `hap_http_debug_enable()` when adding the accessory after an erase, nothing much was added to the logs:
```
Socket fd: 51; HTTP Request POST /pair-setup
######## Starting Pair Setup ########
Pair Setup M1 Received
Pair Setup M2 Successful
I (27864) HAP-GarageDoor: Pairing Started
Socket fd: 51; HTTP Request POST /pair-setup
Pair Setup M3 Received
Using pair-setup without MFi.
Pair Setup M4 Successful
Socket fd: 51; HTTP Request POST /pair-setup
Pair Setup M5 Received
Pair Setup Successful for F4BD5D88-9094-4E49-B869-5E7872DD5759
I (35853) HAP-GarageDoor: Controller F4BD5D88-9094-4E49-B869-5E7872DD5759 Paired. Controller count: 1
Skipping s# update for Certification requirements.
Re-announcing _hap._tcp mDNS service
Cleaning Pair Setup Context
Cleaning Pair Setup Context
```

It still says `Unable to Add Accessory` and that `This accessory cannot be used with HomeKit`.

If i try to re-add after rebooting the accessory, just like before:
```
Keystore initialised
Accessory is Paired with atleast one controller
Updated state number to 2
Database initialised. Accessory Device ID: 26:E9:6B:B6:6A:04
HAP Initialization succeeded. Version : 4.0-3e6955a
I (322) HAP-GarageDoor: Accessory is paired with 1 controllers
```
... with no additional logs.

---
My characteristics are `HAP_CHAR_UUID_CURRENT_DOOR_STATE` and `HAP_CHAR_UUID_TARGET_DOOR_STATE` which are both `int`. However, with `CONFIG_NEWLIB_NANO_FORMAT=n`, I cannot even get the paring to happen:
```
######## Starting Pair Setup ########
Pair Setup M1 Received
Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
Core 0 register dump:
PC      : 0x4010034a  PS      : 0x00000033  A0      : 0x3fff4190  A1      : 0x3fff4190
0x4010034a: _heap_caps_malloc at I:/HomeAutomation/.esp/esp8266/rtos-sdk/components/heap/src/esp_heap_caps.c:174

A2      : 0x3fffb018  A3      : 0x3fffb018  A4      : 0x3fffb234  A5      : 0x3fffb234
A6      : 0x402a54cc  A7      : 0x80000000  A8      : 0x00000003  A9      : 0x3ffefa24
A10     : 0x03101f37  A11     : 0x00000002  A12     : 0x00000001  A13     : 0x00000004  
A14     : 0x3fffb020  A15     : 0x00000214  SAR     : 0x00000010  EXCCAUSE: 0x00000003  

Backtrace: 0x4010034a:0x3fff4190 0x40266825:0x3fff41b0 0x4026caac:0x3fff41d0
0x4010034a: _heap_caps_malloc at I:/HomeAutomation/.esp/esp8266/rtos-sdk/components/heap/src/esp_heap_caps.c:174

0x40266825: ieee80211_deliver_data at ??:?

0x4026caac: sta_input at ??:?
```

Any thoughts?

AMoo-Miki, 2023-10-17:
FYI, I flashed this repo's HEAD with ESP-IDF 5.0's HEAD on an ESP32 and it worked fine.

haecksenwerk, 2023-10-17:
I'm having the same issue with i. e. the smart_outlet example on an ESP8266. I have hardcoded the wifi credentials and the setup-code. A couple of times I succeeded with pairing manually via the setup code instead of using the QR code. But it is not reproducible and I can barely track down something from the log outputs.

BTW: I'm also interested in building a Garage Door Opener. Would you mind sharing your code?

AMoo-Miki, 2023-10-20:
I solved the problem by simply not using `USE_HARDCODED_SETUP_CODE`.

Just in case anyone else needs this:
1. Used `homekit-sdk\tools\factory_nvs_gen\factory_nvs_gen.py <8 digits> <4 letters> factory`
2. Used `idf.py menuconfig` to change the partition to custom and the partition table filename to `partitions_hap.csv`.
3. `idf.py erase_flash` for no good reason
4. `esp8266-sdk\components\esptool_py\esptool\esptool.py -p <port> write_flash 0x340000 factory.bin`
5. `idf.py build flash monitor`

AMoo-Miki, 2023-10-20:
> I'm having the same issue with i. e. the smart_outlet example on an ESP8266. I have hardcoded the wifi credentials and the setup-code.

Try hardcoded WiFi creds but NVS driven setup code following my previous comment.

> BTW: I'm also interested in building a Garage Door Opener. Would you mind sharing your code?

My actual code is complicated by using one normally open and another normally closed contact sensors but the code is just the outlet example switched to use `hap_serv_garage_door_opener_create`.

closed 2023-10-20 by AMoo-Miki